t-oster/VisiCut #216: [Request] Mark Colour

thinkl33t, 2015-01-02:
It would be nice to have a different colour to separate mark and cut operations, similr to the different colour for engrave.  I've seen users get confused which is which while creating their mapping.

t-oster, 2015-01-02:
I don't know if I get your question right, but that should be possible already by creating your own mapping and saving it for other users. You can even set it as default mapping in the options.

t-oster, 2015-01-02:
If you mean a different color in the UI: you can set those in the material properties. Mark-lines are drawn with the "engrave"-color, while cut-lines are drawn with the "cut"-color.

peteruithoven, 2015-01-02:
Currently a cut and engrave color is set per material. I think @thinkl33t is looking for a overriding color per laserprofile. Because then you can differentiate between a mark (which is usually a *line* profile) and a regular cut profile (which is also a *line* profile). 

Currently when you create a laserprofile you get the question whether it's a Line profile, Raster profile or Raster3D profile. I'm assuming the *Line* profile's get the *cut* color and the *Raster* profiles get the *engrave* color of the currently used material?

t-oster, 2015-01-02:
On a "line" profile, you can set the checkbox "is cut", which results in the lines being drawn in "cut"-color. There is also a note next to the checkbox describing that.

t-oster, 2015-01-02:
(you can only find this in the "options->manage laserprofiles" dialog)

peteruithoven, 2015-01-02:
Ah, nice. So a quick fix would be to uncheck that option in the mark profile.

thinkl33t, 2015-01-05:
Aha, that solution is perfect, thanks.

closed 2015-01-05 by thinkl33t