t-oster/VisiCut #221: thingiverse integration

renebohne, 2015-01-13:
A group of students is working on thingiverse integration. 

Main features:
* login to thingiverse
* browse your own things that are supported by lasercutters
* search for things that are supported by lasercutters
* take a photo with VisiCAM and upload it to thingiverse - "I made one"

renebohne, 2015-01-13:
status update:
Visicut now shows the thingiverse login site in an integrated browser dialog. After entering your username and password, the dialog closes and you are logged in.
If JavaFX is not available on your machine, VisiCut opens your system browser instead, and you have to copy/paste a key.
Sessions can be saved persistently. So you only have to login once per computer. Of course, you can logout if you want...

renebohne, 2015-01-13:
status update: Load Image Icons from thingiverse
The problem is that some images which are directly hosted by thingiverse instead of amazon cannot be loaded by java's ImageIcon class. This is the case for fallback user pics, fallback thing pics, etc...

peteruithoven, 2015-01-13:
That makes me very curious, please share some screenshots.

renebohne, 2015-01-13:
EARLY prototype:
[image]

peteruithoven, 2015-01-13:
Nice. Do you search for these lasercuttable designs based on a tag? If so which ones?

tur0kk, 2015-01-13:
update:
- fixed problem with loading images from thingiverse. No fallbacks any more
- providing more user feedback for loading images in the ThingiverseDialog and for loading a thing into VisiCut
- project files can be filtered by lasercutter files (svg, plf, ...)
- show liked things

todo: 
- filter projects also (will be a workaround because not provided by thingiverse)

tur0kk, 2015-01-13:
The problem is that thingiverse do not tag lasercutter things by itself. So it depends on the user who uploads the file. But maybe we can display the ones found by these tags anyway. We will consider that.

Patmachine, 2015-01-20:
We can now filter projects/things by supported file extensions. However, it's quite slow since this is not naturally supported by thingiverse, and we have to do an extra API call for each potential project. Right now a filtered search takes about 4 seconds.

renebohne, 2015-01-22:
general question: can we expect the users to have Java 8? The thingiverse
integration has a fallback for older JREs, but with JavaFX, this new
feature is much more elegant.

t-oster, 2015-01-22:
Hmm... depends. On OS X there is no java installed by default anymore, so I guess people will install the latest automatically. What is default in Ubuntu? I think most Windows installs will also have upgraded themselves to Java 8.

However, VisiCut's official requirement is Java 6

mgmax, 2015-01-22:
Both debian and Ubuntu don't have java 8 in the current release.

Patmachine, 2015-01-24:
Can someone please upload these files (which are needed for the no JavaFX fallback) to the hci website?
https://www.dropbox.com/s/766xpi230fdc823/show_code.css
https://www.dropbox.com/s/alcz8j3g79kj34u/show_code.php

renebohne, 2015-01-24:
Where? Why?

renebohne, 2015-01-26:
done. Here is the URL
http://hci.rwth-aachen.de/public/VisiCut/show_code.php

Patmachine, 2015-01-28:
Thanks!

> Why?
The fallback login now works as follows:
1. Click the thingiverse button. (VisiCut opens your system browser and navigates to the Thingiverse login page)
2. Enter username and password. (The thingiverse website asks you for permission for the app to access your data - only at first login)
3. Accept. (You are now redirected to the above php script - Thingiverse passes the access token as a get parameter and the php script just displays it nicely to the user)
4. Copy/Paste the access token from your browser to VisiCut.

In the preferred login procedure (JavaFX available), the Thingiverse website is instead shown in a browser component within VisiCut and step 4 is omitted.

Icetea1991, 2015-02-02:
[image]

renebohne, 2015-02-04:
The students did a great tutorial video:
http://youtu.be/-YspZ8w3ZM0

peteruithoven, 2015-02-04:
Very cool guys. Would be great if this lowers the threshold for people to share their work.

closed 2015-03-12 by t-oster