t-oster/VisiCut #222: automatic arrangement of shapes

renebohne, 2015-01-13:
A group of students is working on automatic arrangement of shapes.

The main features will be:
* automatic arrangement of multiple shapes on the available material
* detect available material with VisiCAM and avoid unavailable areas

peteruithoven, 2015-01-13:
I would be very interested in the gui designs, maybe we can help? Keeping it simple for first time users for example.

renebohne, 2015-01-13:
The students will read this issue. If you have any ideas and feedback, please use this issue. 
This feature has a deadline: 4.2.2015 :)

t-oster, 2015-01-13:
If the students are allowed to receive help, feel free to frok the project and create a pull-request, even if it is not finished. A pull request is a nice place to colaborate on a new feature.

renebohne, 2015-01-14:
The students are allowed to receive help :)
Part of their final project is to create pull requests to the develop branch.

ksughosh, 2015-02-03:
We have integrated the code for the auto arrangement of objects with VisiCut and it seems to be working. There is a bit of code cleaning that needs to be done before the pull requests. I'll create the pull requests once that is done before or by 3rd of Feb.

t-oster, 2015-02-03:
nice. Please make sure the copyright-header is exactly like in the other files etc. Also please use git-rebase to clean the commit log before creating the pull request, so it looks like strait foreward development ;)

ksughosh, 2015-02-04:
https://github.com/t-oster/VisiCut/pull/237
We have rebased and created pull request.