t-oster/VisiCut #223: facebook integration

renebohne, 2015-01-13:
The same group of students that is working on the thingiverse integration (issue 222), will also provide basic facebook functionality:

* facebook login
* share projects on facebook

it is not clear, how users will benefit from this feature, but maybe other people on github have some ideas :)

tur0kk, 2015-01-13:
Current status:
- facebook login and profile information
- use camera to take a photo
- add a description
- publishing of a facebook post with the taken image and the description
- FabLab location is added to the post
[image]

renebohne, 2015-01-13:
is the fab lab location a new setting in VisiCut? Sounds like a nice general setting to me.

tur0kk, 2015-01-27:
New update:
- We added VisiCam as an alternative source to the webcam (selectable by radio buttons)
- FabLab Facebook Location Id now is a general setting

renebohne, 2015-02-04:
the students also created a tutorial video for this feature:
http://youtu.be/MxeHqAg4TIs

t-oster, 2015-02-25:
Please clean-up the code and commit history before creating a pull-request.

closed 2019-11-01 by mgmax