t-oster/VisiCut #227: Usability Improvements: keyboard shortcuts

renebohne, 2015-01-14:
A group of students will add support for the most important keyboard shortcuts:
command + c, 
command + v, 
delete

t-oster, 2015-01-14:
command+D for duplicate is already there and delete should also work, if I remember correctly.
copy/paste is not implemented, therefore no shortcuts exist.

renebohne, 2015-01-14:
true, but copy and paste will be added with this issue :)
Delete does not work on my Mac.

ksughosh, 2015-02-02:
The command + D option doesn't work. But creating duplicates by right clicking works. However there the new duplicate overlaps on the previous. Hence this occlusion prevents the user from getting the feedback if the part was duplicated. 

It would be better if the duplicated part had some offset from the original so that we could easily distinguish whether the duplicate has been generated. This has been addressed here:

https://github.com/t-oster/VisiCut/pull/236

Also the duplicated part is now on top and is the selectedPart.