t-oster/VisiCut #229: Adding marlin driver for VisiCut

JSilvaGlez, 2015-01-15:
Hello,
My name is Jaime.
Im starting to learn VisiCut source code.
I want add fork marlin driver laser CO2. I need help to start do it.

I hope know about it.

Greats !

renebohne, 2015-01-15:
Marlin needs GCode. 
What you want to do: fork the libLasercut project 
create a new driver (in drivers directory).

you can have a look at our iModela driver, since it also uses a special GCode version :
IModelaMill.java
and add a new property, just like IModelaProperty.java

Cutting Vector paths will be quite easy. Adding support for (bitmap) raster parts, might be tricky!

t-oster, 2015-01-16:
First starting point: https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started
Then read the source of this file: https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/SampleDriver.java

billieblaze, 2015-05-14:
I'd like to +1 this and add, that I've been using the buildlog marlin from here: 
https://github.com/TurnkeyTyranny/buildlog-lasercutter-marlin

would love to get visicut running on this.      looked over the iModela and lasersaur, look pretty tweakable.     

Will trade beer for a driver for marlin!  :)

quillford, 2015-08-19:
@JSilvaGlez did you ever get anywhere with this? I' ve been working on it and have it in an unstable but occasionally working state.

quillford, 2015-09-05:
This can be closed now. See [this](https://github.com/t-oster/LibLaserCut/blob/smoothieboard/src/com/t_oster/liblasercut/drivers/Marlin.java)

t-oster, 2015-09-05:
well, @quillford, looks like  @billieblaze owes you a beer

closed 2015-09-05 by t-oster