t-oster/VisiCut #233: Move all Information to new GitHub page

t-oster, 2015-01-18:
I'd like to move from http://hci.rwth-aachen.de/visicut to the new Github page http://t-oster.github.io/VisiCut/.

Whoever wants to contribute, just create pull-requests on the branch "gh-pages".
I would recommend to clone this into a new repository, since it has nothing in common with the master or develop branch: 

    git clone https://github.com/t-oster/VisiCut.git -b gh-pages --single-branch VisiCutPage

(for write access, replace the URL by your forked git-url.

mgmax, 2015-03-21:
I like the wiki because it is easier to edit. Can't we move as as possible there and only link from the homepage?

t-oster, 2015-04-02:
Of cause. All information, where people might want to change should be in the Wiki. I am more talking about the general project page, as well as "moderated" content like announcements and the download-page.

t-oster, 2017-10-30:
done.

PR #425 (2017-10-30): Merge content of old website into index.html
I collected all information from the old page which I found valuable and added it to the new github page. Fixes #233 (as soon as the old page and visicut.org point to the new github pages page).
closed 2017-10-30 by t-oster