t-oster/VisiCut #243: How do I see the code sended to lasercutter ??

JSilvaGlez, 2015-02-24:
I need study the code sended to lasercutter.
I want learn his functionallity. And will do improves in the future.
I have tried open .plf to read. Its impossible.

Thanks!!

I hope know about it.

Regards

peteruithoven, 2015-02-24:
What kind of lasercutter do you have? You can enable logging to a file in the lasercutter settings.

JSilvaGlez, 2015-02-24:
I Have a Laos Laser. I want learn and improve this project.

peteruithoven, 2015-02-24:
That's always appreciated ;) 
Maybe you can use this:
http://webviewer.laoslaser.org/
It also shows a description on how to enable these output files.

JSilvaGlez, 2015-02-24:
Thanks !! I found the solve.
Now Im studing it 
;)

t-oster, 2015-02-25:
Hi. FYI:
- PLF is just a ZIP file containing the source graphic files (svg,png etc.) and some xml files containing VisiCut's settings. You can just rename it to .zip and then open it. It does not contain any information on the lasercutter, nor the code which is sent to it, because that code is generated on the fly once the lasercutter has been selected in VisiCut.
- If you want to know how to control a LAOS-Laser, read the LAOS wiki about simplecode http://redmine.laoslaser.org/projects/laos/wiki/SimpleCode or use the "debug-file" setting in the lasercutter preferences, so every output is also saved as a file
- If you want to know how VisiCut sends information to Lasercutters, have a look at LibLaserCut's drivers: https://github.com/t-oster/LibLaserCut/tree/develop/src/com/t_oster/liblasercut/drivers
- If an issue is solved, please close it.

closed 2015-02-25 by t-oster

peteruithoven, 2015-02-25:
That's very useful information, I added it to the wiki. 
https://github.com/t-oster/VisiCut/wiki/PLF-format
https://github.com/t-oster/VisiCut/wiki/LAOS-Laser-driver

JSilvaGlez, 2015-02-26:
closed issue #243