t-oster/VisiCut #252: UTF8 characters in default mapping doesn't work under Win7

mgmax, 2015-03-21:
This bug is not so easy to reproduce. I suppose it is like this:

Loading the FAU FabLab settings, that were created under Linux, seems to work fine. (default mapping is "schneide rot ..." which contains an umlaut)
But manually setting the preferences (or even saving them unmodified) causes them to break.

Is there some charset issue?

I will test this further...

mgmax, 2015-03-21:
At least under WinServer2008 the umlaut is converted to a questionmark when saving the preferences. Under Linux this is not the case.

t-oster, 2015-04-02:
It may be, that filenames in Windows are saved in Windows-1252 instead of UTF8? IMHO in the java runtime, strings are always unicode, so I suggest that the file-names when written to disc are to blame?

commit b74bd16f (2018-12-27): Fix problems with special characters in settings

Change all XStream serialization to UTF8.

Fixes #252
closed 2018-12-27 by t-oster