t-oster/VisiCut #256: HPGL Generic Driver

DatanoiseTV, 2015-03-31:
I would like to request a generic HPGL Driver Example to be used for HPGL-Compatible lasers. This could be used to extend VisiCut to be compatible with a wider range of industrial lasers.

t-oster, 2015-03-31:
Hi,
drivers are implemented in LibLaserCut. Look at the Epilog-drivers, they use a HPGL subset for the vector part (https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/EpilogCutter.java#L655) . I think there is no gerenic HPGL subset suitable for lasercutters, because there are no parameters like Power, Speed and Frequency in HPGL?

The next problem is the transport layer. What do you suggest to send the HPGL code to? Ethernet? TCP/IP? LPD (the one the Epilog drivers use, but empedded in PCL). Serial Port? Parallel Port.

Can you give a few examples of Lasercutters which understand HPGL?

DatanoiseTV, 2015-03-31:
Hi,

I would like to provide support for USB-Connections (via libusb) and Ethernet via TCP/IP. One could think of an embedded device (RPi2 for example) as a TCP/IP to USB Bridge eventually if implementing libusb support in Java is too difficult (I'm more a Python / C / C++ guy).

The certain family of lasers I am thinking of are the Trotec (Speedy) family.
http://www.seri-deco.fi/files/Datasheet_8018_SP1500_english.pdf

t-oster, 2015-04-02:
I already talked with some people from Trotec. They did seem quite interested on VisiCut. Unfortunately I do not (yet) have access to a Trotec device in order to do any tested development. I think libusb may be possible, but I have no experience with it yet, so I do not know how the Trotec lasers work.

Do you have any specifications on the HPGL subset, they understand and how exactly it has to be transmitted?

Did you look into the Epliog driver?

kr15h, 2015-07-03:
Come to FabLab Berlin, we have a Trotec for you to tinker with. We also need that driver for VisiCut.

t-oster, 2015-07-03:
@kr15h I'd like to come and do some hacking, unfortunately my schedule is always full since I am not a student anymore and thus I have much to less time for VisiCut. But I agree that a driver for Trotec devices would be a huge benefit for VisiCut.

mgmax, 2018-02-25:
@DatanoiseTV This was fixed with the "GoldCutHPGL" driver, which should be generic.

mgmax, 2018-09-25:
@t-oster please close this issue, it was fixed with the "GoldCutHPGL" driver. Thanks.

closed 2018-09-25 by t-oster