t-oster/VisiCut #262: UI too wide for 1280x1024

mgmax, 2015-04-15:
on an average old TFT with 1024 pixels width, the UI does not fit and the "submit" button is clipped.
I tested this in german under win7 and debian, I measured a minimum width of 1360 pixels.

adnbr, 2015-05-05:
Also too wide for 1024x768, thanks.

mgmax, 2015-05-05:
I did some promising experiments with GridBagLayout, but they are not 
yet finished

mgmax, 2015-05-05:
I'm giving up for now, swing's method of determining preferred widths is a horrible crap.
I nicely stuffed the components of the right panel into a GridBagLayout, which works rather nice. But still the whole application has a default width of 1220px. Aaargh!
a JSplitPane would be cool, but its resizing does not work as expected, java still thinks the window is overflowing...

maybe the problem is also in PreviewPanel...

If anyone wants to play around with this, my latest code is in an experimental branch my repo (just follow the link in this issue)

I tried this netbeans 7.3.1 with the swing plugin http://plugins.netbeans.org/plugin/43853/swing-application-framework-support.
older versions have broken code generation for GridBagLayout, newer versions don't support our swing GUI anymore (except with the plugin, not yet tested on any newer version).

mgmax, 2015-05-17:
I made it work with 1280 using netbeans 7.3.1 + plugin. 1024 width was not possible, that is just too small and I think it is simply not possible with the current layout and framework.
I published this in my repo.

Can someone test this on MacOS? We had lots of trouble with MacOS before, so this needs to be tested before being pulled in.

closed 2015-05-20 by t-oster

FroChr123, 2015-05-31:
What do you think of just adding a second top navigation row and adding some more vertical space there?

As it seems the main issue is the horizontal space. Do I understand this change correctly, that there will be no possibility to use these buttons on smaller displays with this change?

Unfortunately I need to add at least one button for my addition to this project, but I might be able to remove that huge "Capture photo" button.

mgmax, 2015-06-02:
I didn't mean to hurt anyone with this commit: Before you just could not 
use VisiCut at all on low resolutions, except for the social buttons 
that need at least 1450 pixels width.
The "capture photo" button is necessary and should be large enough to be 
clearly visible.

Usually horizontal space is not the large problem, screens will 
constantly become wider, but not higher - most higher end notebooks have 
1920x1080, which is mostly the same height as 1280x1024.

I know it is a matter of taste, but in my opinion we should focus more 
on making the basic operations work better, rather than adding lots of 
features.

peteruithoven, 2015-06-02:
I agree with @mgmax. It's like designing a responsive website, you can't just add everything from the widescreen website into the mobile dimentions. 
Can't we add these features to the native top left menu bar?

FroChr123, 2015-06-02:
I am just looking for the best solution for this issue, which is the reason why I want to discuss it a bit with you. :)

I have also noticed that this additional stuff takes up too much horizontal space because of the large amount of buttons, especially on my laptop display.

But at least for my resolutions and settings, only 50% of the left half of the screen are used, if the preview is set to zoomlevel "fit window". This is the reason why I came up with the idea to use this unused and available vertical space for a second row of elements.

Do you have other settings or do you usually use higher zoomlevels? For these cases a second row would indeed not be a good solution again. But I also think that the "disappearing" of UI elements should be avoided, for me it does not feel right to remove the possibility for the user to perform some actions because of a small display.

Maybe the idea of @peteruithoven is a good solution, maybe we can move some stuff up in the main menu bar to save some space.

For the photo button I am currently thinking of changing it to a checkbox, which is activated by default and gets a photo frequently, if possible. With this change I was already able to get a near-to-real-time video as background images there.

commit 0bde1c5d (2015-05-20): Merge pull request #274 from mgmax/tmp-262-5

Fix GUI resizing. Fixes #262
commit ad8606d8 (2015-05-20): Fix GUI resizing. Fixes #262
now works down to 1024x768 except for facebook/thingiverse buttons

Changes:
- put zoom and other buttons inside a toolbar so that the rightmost ones disappear when there is not enough space
- remove some spacing
- changed resizability and default width attributes at a few places

done with netbeans 7.3.1 + swing plugin

tested on windows and linux