t-oster/VisiCut #263: Compile issue OSX

DatanoiseTV, 2015-04-15:
The build with "ant" fails with:

BUILD FAILED
/Users/datanoise/Desktop/VisiCut/nbproject/build-impl.xml:931: The following error occurred while executing this line:
/Users/datanoise/Desktop/VisiCut/nbproject/build-impl.xml:303: Compile failed; see the compiler error output for details.

Total time: 4 seconds

t-oster, 2015-04-15:
can you get more lines? which jdk is installed?

peteruithoven, 2015-04-15:
I think I have the same issue: https://github.com/t-oster/VisiCut/pull/260#issuecomment-91844052
How do I check the jdk version? `java -version`?
```
$ java -version
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-462-11M4609)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-462, mixed mode)
```

t-oster, 2015-04-15:
The thingiverse stuff uses newer java 8 api.can you upgrade?java 6 is old anyway

I think I have the same issue: #260 (comment)

mgmax, 2019-11-01:
Closing because the report refers to a very old version. Feel free to reply if the bug also happens in a recent version of VisiCut with Java >= 9.

closed 2019-11-01 by mgmax