t-oster/VisiCut #265: svg causes IllegalArgumentException

mgmax, 2015-04-23:
This file https://gist.github.com/mgmax/46627e2cda48937da772 causes a strange crash in VisiCut.
```
Exception in thread "AWT-EventQueue-0" java.lang.IllegalArgumentException: User must specify at least 2 colors
        at com.kitfox.svg.batik.MultipleGradientPaint.<init>(MultipleGradientPaint.java:126)
        at com.kitfox.svg.batik.LinearGradientPaint.<init>(LinearGradientPaint.java:259)
        at com.kitfox.svg.LinearGradient.getPaint(LinearGradient.java:123)
        at com.kitfox.svg.ShapeElement.renderShape(ShapeElement.java:163)
        at com.kitfox.svg.Path.render(Path.java:92)
        at com.t_oster.visicut.model.graphicelements.svgsupport.SVGObject.render(SVGObject.java:121)
        at com.t_oster.visicut.gui.beans.PreviewPanel.renderOriginalImage(PreviewPanel.java:465)
        at com.t_oster.visicut.gui.beans.PreviewPanel.paintComponent(PreviewPanel.java:587)
...
```

mgmax, 2015-04-23:
On Windows I thought VisiCut was broken, because it does not show any error message. I had to open it in the shell on Linux to see the backtrace.

renebohne, 2016-02-03:
I opened the svg in 1.7.9 (Windows 10) and I can't see any problems. When does it crash?

mgmax, 2016-02-03:
Hmpf, there are errors on the commandline with latest VisiCut, although they don't seem to cause trouble - or I don't remember anymore how to trigger the problem. VisiCut behaved strangely, probably because some drawing/UI operations were halted by the exceptions. It seems this is not the case anymore.
I don't feel good about closing this issue, we should rather add a try/catch somewhere

mgmax, 2016-02-03:
Okay, looking at the output again, it was already caught. Fixed by https://github.com/t-oster/VisiCut/commit/4b7205dc7de731b23e44a1f7e086c71368e10d20 , closing, thanks and goodbye :-)

closed 2016-02-03 by mgmax