t-oster/VisiCut #269: Engrave squashed but cuts correct

iMartyn, 2015-05-09:
We use a LAOS retrofitted chinese copy of a HPC6040 and the cuts are working perfectly.

Unfortunately when we send an engrave job, it is squashed by about 93%.

Given that the steps per mm is right for cuts, I don't understand how it's not for engraving?

If it's a different setting, I'm not finding it, but potentially it could be useful to have a scale factor for engraving in the gui?

mgmax, 2015-05-09:
I don't have a LAOS device, so it is difficult to check whether your device or VisiCut is at fault.
What you could do is:
- try to reproduce the problem with a testcase as simple as possible, e.g. a 5x5 cm square that is engraved grey and then cut out
- Please make a VisiCut screenshot and a photo of the lasercut result. If possible, also measure it as precisely as possible so that we can find the scaling factor.
- You can save a debug file and show it in the web viewer, see  https://github.com/t-oster/VisiCut/wiki/LAOS-Laser-driver (see there for instructions how to get  the SimpleCode file). unfortunately, the web viewer seems half-broken for me and does not work with the chromium browser.
- upload your test file somewhere

Does someone know where to report bugs in the LAOS web viewer? Neither the site nor the sourcecode states any author.

@peteruithoven Do you have a LAOS machine to test on, or a sample SimpleCode file with correct engraving?

iMartyn, 2015-05-13:
Okay cool : 
https://cloud.ranyard.info/public.php?service=files&t=f47f8fe366bd1a30135cc62310b0c966 is a 5x5 next to a 5x5 square.  Isn't really obvious there, so...
https://cloud.ranyard.info/public.php?service=files&t=6345d977052042aa84beb239b55e0294 is a 20x20 + 20x20 and calipers.

The debug file is https://cloud.ranyard.info/public.php?service=files&t=7a064c1acbd6b52b1fe662876350e486 and in firefox it does look the same height as the cut :(

https://cloud.ranyard.info/public.php?service=files&t=223438d22883c65ff2f75fd8e80314e9 is the screenshot of visicut.

The measurement of the height is 17.83 interior where the width which is right is 18.72 (which accounting for kerf is about right)

@thinkl33t could you try the simplecode on the manchester laser or send me a similar test file?

just for completeness, https://cloud.ranyard.info/public.php?service=files&t=d26f78f985e85c9d88bfa8ec19aed321 is the svg.

Let me know if there's anything else I can provide.

iMartyn, 2015-05-20:
Upon further investigation, we have more information:

* This effect on the etch only happens if you run bidirectional etching.
* We are seeing some very strange results when cutting certain complicated shapes as well, for instance a gear that should be circular ends up spiral.

Given the machine cuts simple cuts perfectly, someone suggested that perhaps it is due to acceleration/deceleration, perhaps that when it is decelerating X, Y is not moving as far?  Just a thought though, nothing clear enough to have a test case for.

Any suggestions welcomed.

t-oster, 2015-05-20:
I bet it is some speeds in the LAOS config set too fast, so the stepper motor "skips" some steps depending on current accelleration and direction.
Did you try cutting/engraving the same part with 100% and 2% speed, to see a difference?
Have a look at your LAOS config file and see the values for motion (http://redmine.laoslaser.org/projects/laos/wiki/LAOS_configuration_file) i.e. 
motion.speed  and motion.accel

peteruithoven, 2015-05-22:
I could give your files a try on Tuesday or Wednesday (26, 27 may). 

You could also attempt to debug it using the laos simplecode webviewer. 
http://webviewer.laoslaser.org/
Although it currently lacks proper engrave support, see: 
http://redmine.laoslaser.org/boards/3/topics/640
Maybe you could help improve it?

thinkl33t, 2015-06-03:
@iMartyn I think when we set up your laser, we diddn't tweak the acceleration or the current potentiometers on the pololus, and we imported the config from my laser, just scaled up.

It might be that the acceleration that's set on there is too fast for the combo of motor current and bed width, I'd try either decreasing the acceleration or increasing the stepper current.

I should have the new Manchester laser (the same model as yours) up and running by the weekend, so will be able to test your files on the same size machine.

mgmax, 2015-09-05:
@iMartyn Can this issue be closed, as it is not in VisiCut?

iMartyn, 2015-09-06:
Sure can, it's not visicut at fault I agree.

closed 2015-09-06 by iMartyn