t-oster/VisiCut #271: Usability: multiple colors, same profile, wrong expectations

renebohne, 2015-05-13:
This is not a bug, but something that most visitors get wrong: they use different colors and want to say that one color uses different settings than another color. In die example, the user selects "mark" for both colors and then switches to the "Laser Settings" tab. He only sees mark settings for "black" - he thinks that the other color is missing.

How can we offer a way, to add a new profile for the second color directly in the tabs? The user probably wants to create a profile "mark2" and just change the power or speed parameters a little bit...

And yes, most users with this question have about 10 different colors in their svg file.

Here is a video that shows the situation (click on the image):
[image]

peteruithoven, 2015-05-17:
Hi Rene,
I totally get the issue, I've been having issues with the mapping and settings interface for quite a while, we did brainstorm om some designs, but never implemented a overhaul. See the following issue: 
Did you see the following ui brainstorm? https://github.com/t-oster/VisiCut/issues/168

But, the following issue also seems relevant: https://github.com/t-oster/VisiCut/issues/232

t-oster, 2015-12-16:
is this #319? Then you should maybe close it.

thinkl33t, 2015-12-16:
I think this issue specifically refers to splitting out multiple toolpaths of the same type (say, two engraving operations) into different power / speed combinations, whereas #319 is just adding UI hints to show what toolpath applies to what colours.

mgmax, 2016-05-07:
The underlying problem is that we're addressing two distinctively different levels of complexity here. Let me explain it by two "user stories".
- Joe Average just wants the result of an operation (cut/engrave/mark/..). For this he uses the predefined settings found in the lab. VisiCut correctly hides the underlying details from him, just like classes in object-oriented programming hide the inner workings.
- Sandra Specialist wants to creatively (ab)use the lasercutter by semi-cutting, bending, marking with different depths, engraving with different dithering settings in one file and whatnot. In object-oriented programming, she would use a lower class and directly use all its methods. In a GUI it is really difficult to satisfy her wishes without scaring away Joe Average.

Sandra Specialist might be looking for something like the (in my opinion horrible) Epilog Windows driver with thousands of buttons and sliders:
[image]
[image]

It is quite difficult to bring these two worlds together, and my personal opinion is that we should be careful not to loose Joe Average. In our lab, we give users the choice between VisiCut and the Epilog driver and most people choose VisiCut because they feel it is easier. This is an achievement which was hard to gain and should not be given up easily.