t-oster/VisiCut #272: Correction of kerf size [Feature Request]

Koshu, 2015-05-16:
I would like to know if there are plans to implement a kerf correction by growing or shrinking the cut paths. Adding it as parameter to visicut would have the benefit to allow adjustments for different materials  without the need to change the svgs (dxfs ...) every time. 

I would be willing to help implementing this feature.

Greets,
koshu

hpux735, 2015-05-16:
Wouldn’t it be virtually impossible for the program to know which way move the line?

Koshu, 2015-05-16:
Well, it needs to know if the line is an outline or inside the manufactured part. This probably has to be set manually in VisiCut or must be encoded somehow inside the svg (i.e. some online laser cutting companies use colorcodes). Beside that only the size of the kerf needs to be known for the given material, which is normally determined experimentally anyway.

Koshu, 2015-05-16:
Fiddled around a bit with the sourcecode. Still needs a option for setting inside/outside and some optimizations, thought.
[image]

hpux735, 2015-05-16:
The problem will come from complicated shapes where not everything is a closed polygon, I imagine.

Koshu, 2015-05-16:
Open shapes are an issue, but then again most cases where a kerf correction is useful are for the dimensions of your parts, especially if your are building something from multiple parts and they should fit together. Beside a few rare cases i can think of (maybe when you want to bend something), all of them are closed shapes or could be converted to closed shapes. Or am i missing something?

hpux735, 2015-05-16:
This one comes immediately to mind...

Koshu, 2015-05-16:
Was there an attachment in your reply? If yes, I can't see it.

hpux735, 2015-05-16:
http://www.thingiverse.com/thing:6223

It looks like their site is defunct...

Koshu, 2015-05-16:
Ah ok. But these part could still be designed as a closed polygon. Its just that they are placed in a most inconvenient way to be completely "kerf-corrected" because even if done manually you can't without changing the other part. I guess you could add some kind of priority system to mark which lines have a function and need to be precise but that sounds a bit over the top. In most cases you could just add a little bit of space between the parts (in your case it's a bit of a problem, since 3 sides are touching at the same time).

hpux735, 2015-05-16:
I agree that it’s possible to make it work in most cases.  It will require the user to keep this step in mind when designing their graphics.  But, it care is taken early then it could definitely pay off.

Koshu, 2015-05-16:
At least i hope so. After getting netbeans 7.0 to run i was able to add some controls. For now I added it to the laser profile options. While I will probably delete most of the lines i added for debugging, i plan to give the user some kind of visual feedback of what will happen, so overlapping lines etc are noticed. I will create a proper fork for this feature after some code cleaning.
[image]

Koshu, 2015-05-17:
For anyone interested and maybe want to help/test: 

https://github.com/Koshu/VisiCut/tree/feature_kerfCorrection

computercarguy, 2016-06-27:
Any updates on the status of this very nice feature?  I have more than one project that I've created in CAD and have not accounted for kerf.  I hate to think about redoing these projects because of a less than a mm correction.  I went to the branch, but didn't notice any status update as to code complete or what.  Thanks in advance!

Koshu, 2016-06-27:
In theory the code in my github should work. Te UI has some glitches and isn't that user friendly but the algorithms are in place. However, for some time a didn't have a chance to do some in field tests, so I never really tested the code. So, if you are brave enough and and don't need the newest visicut give it a try (no warranty, of course).