t-oster/VisiCut #277: Inkscape Plugin: Execute with current Mapping

t-oster, 2015-06-08:
Since I am using Inkscape a lot to prepare my lasercutter jobs, I'd like to extend the inkscape extension in a way, that it will send the currently selected elements relative to the document coordinate system and directly executes it with the currently selected mapping in visicut.

The workflow would then be: Create a document in inkscape (probably with the dimensions of the lasercutter), hit "open-in-visicut" to apply a mapping and maybe set a position etc. Then on further iterations, just manipulate the drawing in inscape and select "execute in visicut", which then reloads the file, re-applies the mapping and transform and sends the job to the lasercutter without even popping up.

As soon as I have time, I will implement this. Suggestions welcome.

tur0kk, 2015-12-29:
I once saw someone implementing something very similar in Java. He solved it by observing the timestamp of the opened file in a background thread, and on change, reload it. 

Wouldn't that be an easy solution here? Then one can just save the modified file in Inkscape and VisiCut reloads the changed file.

mgmax, 2015-12-29:
Unfortunately no. The inkscape plugin does some cleanup before sending 
the file to VisiCut. It does not know the original filename because 
inkscape copies to a new temporary file with different name every time 
you use a plugin.