t-oster/VisiCut #279: Error refreshing picture: null even if camera is disabled

t-oster, 2015-06-19:
I think since I merged the changes for #276, I get an error whenever I open a file, which says that the camera picture could not be refreshed. In my settings, the lasercutter does not have a camera URL so camera should be disabled.

FroChr123, 2015-06-19:
Hi, hm that is strange. I am currently unable to reproduce what you described with that version.

I tried the following:
1) Empty camera URL - Restart VisiCut - Loaded file. No message.
2) Enter invalid camera URL - Of course error message - Clear camera URL - Loaded file. No new message.

The first thing, which confuses me a bit: Are you sure, that it is related to loading new files? The only occurence of captureImage should be in the class RefreshCameraThread, which should not be changed in that way, which would cause such messages, when new files are loaded.

Currently I can not find a reason for this behavior in the code (on this version) as well. Related code parts:

```
public String getVisiCam()
{
    LaserDevice dev = MainView.this.visicutModel1.getSelectedLaserDevice();
    return (dev != null && dev.getCameraURL() != null) ? dev.getCameraURL() : "";
}

// Call capture new image if everyhing is fine
if (!MainView.getInstance().getVisiCam().isEmpty())
{
    MainView.getInstance().captureImage();
}
```

mgmax, 2015-08-29:
The exception class name was not printed in certain cases, which will be fixed by #296 ."null" was probably just the .getLocalizedMessage().
Also, you should see a stack trace on stderr.

mgmax, 2015-09-05:
@t-oster I think the bug can be closed now because either it was something else or it has accidently been fixed :)

t-oster, 2015-09-07:
I cannot reproduce it anymore, so I consider this as fixed.

closed 2015-09-07 by t-oster