t-oster/VisiCut #280: Add option for low-resolution preview

t-oster, 2015-06-19:
When I designed VisiCut, I had accuracy in mind and I created the preview of raster/engrave-mappings in a way, that there is a pixel for every possible laser "dot". So, depending on the stepper-motor resoution (500dpi on an epilog zing) the preview images for raster-mappings become quite large, which explains the long rendering time.

I did this, because I wanted to have an accurate preview on how the dithering-alogrithm (which converts the color-image into black-white pixels) affects the image. However I think in most cases, users won't mind the difference, so I think it would be a good performance boost, if we render the preview only in the resolution of the current preview size, or at screen size to not have to re-render it on zooming.

I would add an option like "fast-preview (less acurate)", so the users who need accurate preview can still enable it.

mgmax, 2015-06-19:
How about switching it dependant on the current zoom level: If you zoom 
out far enough you cannot see it anyway, and currently we have lots of 
ugly aliasing in the preview, so we can also show a low-res one. If you 
zoom in near to a 1:1 pixel preview, the lowres one would be confusing, 
so show the real preview.

t-oster, 2015-06-24:
nice idea. That's why I first write down my ideas to github, so I can get feedback before implementing ;).  The question is how to handle changes in zoom level. If we re-render on every step, it may be time consuming, when the user is zooming in and out quickly.

Maybe we always use the last rendered version enlarged or shrinked as preview and if the zoom level stays constant a certain timeout (e.g. 3s) we start rendering.

mgmax, 2019-11-01:
This has now mostly been implemented (depending on the zoom level, you get more or less resolution in the preview).

mgmax, 2020-08-16:
Closing as it's mostly fixed (except for the render time)

closed 2020-08-16 by mgmax