t-oster/VisiCut #284: inkscape cloned groups fail to show all objects

jnweiger, 2015-07-12:
Repeat this in inkscape:

draw two rectangles. Group them. Create a clone of the group, rotate the clone 90 deg, save.
Open in VisiCut, I see only 3 rectangle, not 4.

jnweiger, 2015-07-12:
box_cloned.svg https://gist.github.com/jnweiger/12ca918954676cd3b25c

mgmax, 2015-07-12:
Hi Jürgen :)
Using the inkscape plugin everything works for me. (as expected, because it unlinks clones)
Directly opening the SVG in VisiCut fails to cut, but raster-engrave works.
So the trouble must be at the point where cutting paths are determined from the SVG.

matthijskooijman, 2020-08-16:
This is still an issue in the latest visicut (1.9-70).

Thanks for the tip about using the inkscape plugin, that indeed seems to work around this problem.

cdemeijer, 2020-10-06:
I suspect that #354 may be related. Seeing as both use references within the file.
Still trying to wrap my head around this one though.