t-oster/VisiCut #286: Illustrator extension for Windows

diegohuyke, 2015-07-21:
I think it would be really useful to have an Illustrator extension that also works on Windows. Currently, because it's written in AppleScript it only works on Macs. So I thought VBScript would be easy to make it work on Windows.

I don't really know a lot about programming so I worked on the script itself using some Adobe documention but not on the installer through VisiCut. Like how to install the Inkscape extension you go to Extras>Install Inkscape-Extension. Anyways, here is the code for the script:

     'Exports the current document as SVG to dest
     'dest contains the full path and file name to save to

     destFile = "C:\temp\VCtemp.svg"
     exportFileAsSVG (destFile)

     Sub exportFileAsSVG (dest)
          Set appRef = CreateObject("Illustrator.Application")
          Set svgExportOptions = CreateObject("Illustrator.ExportOptionsSVG")
 
          If appRef.Documents.Count > 0 Then
               svgExportOptions.EmbedRasterImages = True
               svgExportOptions.FontSubsetting = 7 'aiAllGlyphs
               Set docRef = appRef.ActiveDocument
               Call docRef.Export (dest, 3, svgExportOptions) ' 3 = aiSVG
          End If
     End Sub

     'open document in VisiCut

     strFileName = "C:\temp\VCtemp.svg"
     Set oShell = CreateObject("WScript.Shell")

     oShell.Run "visicut.exe "  & strFileName

If you save this code as a .vbs and place it in your Illustrator's \presets\scripts folder it should work, but I'd like to be able to install it like you do with the Inkscape extension. After that you could just select the extension from the scripts menu of Illustrator.

On NetBeans, I've been looking into the Helper.java file where I find the instructions for Mac Installation. So what's my next step?

t-oster, 2015-07-21:
Hi, the next step is:
 * Put your script in the directory tools/illustrator_script directory https://github.com/t-oster/VisiCut/tree/develop/tools/illustrator_script
 * Modify the "install Illustrator Script" method https://github.com/t-oster/VisiCut/blob/develop/src/com/t_oster/visicut/misc/Helper.java#L303 to check the OS like in the installInkscapeExtensionMethod https://github.com/t-oster/VisiCut/blob/develop/src/com/t_oster/visicut/misc/Helper.java#L249, and if it finds Windows, it should put the script in the right path
 * Create a clean pull-request (single commit) from the develop branch for me to test and integrate it into the distribution

diegohuyke, 2015-07-21:
My problem is that I have no idea what I'm doing so I started by making the code work with only Windows as a starting point. I did steps (1) and (3) of you post: I changed the script name in line 661 to just the windows-specific script and I replaced `isMacOS()` with `isWindows()` in line 666.

For step (2),  I simply added another new File directory, but this hasn't done anything.

      for (File dir : new File[]{
           new File("/Applications/Adobe Illustrator CS3/Presets"),
           new File("/Applications/Adobe Illustrator CS4/Presets"),
           new File("/Applications/Adobe Illustrator CS5/Presets"),
           new File("/Applications/Adobe Illustrator CS6/Presets"),      
           new File("/Applications/Adobe Illustrator CS4/Presets.localized"),
           new File("/Applications/Adobe Illustrator CS5/Presets.localized"),
           new File("C:/Program Files/Adobe/Adobe Illustrator CC 2015/Presets.localized"),
         })

At this point, when I open VisiCut, I'm able to select the `Install Illustrator-Extension...` option. When I select it, it says that it was successful but the script is not installed. Listing the exact location (without the .localized) doesn't help either. What am I missing, does adding a new directory like that not work for Windows?

Thanks

t-oster, 2015-07-21:
I think it is only working, if you build VisiCut to a package, not run it out of netbeans. You should however get a kind of error message or exception in the netbeans console

t-oster, 2015-07-21:
I created a pull request (#287) for you. I made all the changes I think are necessary, and I will upload a build for you to test. Since I dont have windows and illustrator, I cannot test it. Please continue discussion on #287

closed 2015-07-21 by t-oster