t-oster/VisiCut #288: Generic Gcode driver (smoothieboard, grbl, marlin)

t-oster, 2015-07-24:
I am working on Support for Gcode based lasercutters. Feedback welcome. Plase here https://github.com/t-oster/LibLaserCut/pull/23

freakyattic, 2015-10-26:
Dear t-oster,

I am working in a generic Gcode driver for GCode interpreters such as Mach3, LinuxCNC or Marling firmware. I have discover after I start to code the driver, that you have made some developments for a GCode driver, I just forgot to look into the Smoothieboard code branch.

So far I have implemented the Cut, Mark, Raster and Raster 3D, and tested them visually analyzing the GCode Output. The lens in my laser got dirty and destroyed, I am waiting for a new lens to run the tests.

When I am satisfied with the results, If you want I can issue a pull to the main project, so you can add it. I may change the driver name, because I have used the same name as yours.

Thank you for this nice program and your help to the community.

My repo: https://github.com/freakyattic/LibLaserCut

t-oster, 2015-12-16:
The generic GCode driver made it into the develop branch. It is not fully tested yet, but should work in most cases.

closed 2015-12-16 by t-oster