t-oster/VisiCut #289: [request] Save output file directly.

thinkl33t, 2015-07-27:
It would be handy when using the LAOS or smoothie drivers within LAOS to have a button that allows the user to easily save the file for later usage / testing.

I know this is possible within LAOS using the debug output, but i was thinking popping up a Save As dialog would be a bit easier to use.  

UI wise, I think there is room for a button to the left of the execute button, which would only be visible if the current laser's driver supports file export.

peteruithoven, 2015-07-27:
I'd also like for this to be easier. But I wouldn't add a button for this in the main interface. I'm afraid users might confuse it with a save button for example. 
What about adding an item to the *File* menu? Something like *Export output file ...*?

thinkl33t, 2015-07-27:
That sounds like a good option.

t-oster, 2015-07-27:
hmm... I think there was already a similar request/discussion with jaap. definitely a good option, should also be not hard to implement.... time is the only limiting factor ;)

t-oster, 2015-11-25:
I found the other request: https://github.com/t-oster/LibLaserCut/issues/18

t-oster, 2015-12-16:
this was implemented in #316

closed 2015-12-16 by t-oster