t-oster/VisiCut #290: Log

quillford, 2015-08-05:
Is there a log of what was sent to the laser cutter?

peteruithoven, 2015-08-05:
See for example: 
https://github.com/t-oster/VisiCut/issues/243
https://github.com/t-oster/VisiCut/issues/289

closed 2015-08-05 by quillford

valkyriesavage, 2023-03-06:
I don't see the functions described in the various places around the wiki/issues areas that actually allow saving.
* wiki [suggests enabling debug file output](https://github.com/t-oster/VisiCut/wiki/VisiCut-manual#lasercutter-properties), but this is not an option (I see only hostname, port, bedwidth, bedheight, autofocus, softwarefocusnotsupported, type, origin bottom left, job prefix, job sent text, name, and description on visicut 1.9-141-ge75914f3)
* issue #243 is only about a LAOS laser, as far as I can tell. I have an Epilog Helix which doens't have those settings. From what I can tell of the [service manual](https://metalab.at/wiki/images/d/dc/EpilogServiceManual.pdf), the laser itself does not log the files anywhere (it doesn't really have enough memory)
* the export laser code button in #289 is helpful, but I need an automated solution so that I can track overall usage of the laser in a large fablab

t-oster, 2023-03-06:
In normal operation visicut does not log files sent to the lasercutter. You could create a proxy which intercepts the data sent to the lasercutter from visicut, dumps them to a file (or does whatever you want) and then forwards it to the real lasercutter.