t-oster/VisiCut #291: VisiCam auto refresh: keep old picture on error

sedrubal, 2015-08-08:
I tried visicut 00eb2eb with visicam and realized the automatic photo refresh every 2 (?) seconds.

When I close the lid on our lasercutter, the camera can't take a picture anymore. After a short time, visicut looked like this: :smile: 

[image]

I'd suggest to keep the old picture when automatically refreshing if there is an error...

mgmax, 2015-08-10:
I suggest that the image should rather be hidden or greyed out, so that the user notices it is no longer up to date.

sedrubal, 2015-08-10:
It might be also a good idea to disable this feature because it's more confusing than useful.

peteruithoven, 2015-08-10:
> It might be also a good idea to disable this feature because it's more confusing than useful.
Excuse me? I believe that depends on the context? It's very fresh so it's bound to be imperfect, especially with all the different camera setups. 
How come you're camera stops working when the lid is closed?

sedrubal, 2015-08-10:
Sorry, I think, this sounds quite unfriendly vut I like your work and the time you spend on this project, but in my opinion an automatic refresh is not really useful, because the continuously changing background makes it hard to concentrate on the model. And if an image is wrongly clipped, it is jumping around.

Our camera stops working, because we use [this](https://github.com/t-oster/VisiCam/wiki/Cameo-Zing-6030---Logitech-Pro-9000-Mount) construction.

peteruithoven, 2015-08-10:
But how come the camera stops responding when you use that construction? Is it temporarily turned off somehow when the lid is down?

sedrubal, 2015-08-10:
No, but visicam doesn't find the marker and so the image is invalid

sedrubal, 2015-08-20:
Now I tried ef61ca and I like the camera button. But why did you remove it in version 0eb2eb?

t-oster, 2015-08-21:
I think there was a large discussion on #276 ... however, I just merged the code, I did not remove the button myself.

mgmax, 2015-09-05:
@sedrubal can you test the current version, as it is running in the lab, and report back? I don't really see the use of having an outdated picture -- this has confused users sometimes before.

sedrubal, 2015-09-06:
As visicut has no reasonable releasing manner, I can't say if it's the same version, but in `1.7-256-g16bd3fb` it's much better, thank you. But I still miss the "auto refresh: Off" button.

Btw. I don't think that an picture is outdated, only if the camera can't take a newer one...

mgmax, 2016-05-07:
@sedrubal looking at the typical workflow of Visicut and Visicam, a picture is outdated with very high probability if the camera cannot take a newer one:

1. Switch on lasercutter, open lid, put in material
2. load SVG in Visicut, position on material
3. send to laser
4. close lasercutter lid, press start. Now capturing an image may (depending on the camera and marker setup used) no longer be possible. The old image still shows the uncut material, which is no longer correct.
5. Open lasercutter lid, now VisiCam will take correct pictures again.

(remark: Sporadic single-frame capture errors are not a problem here because VisiCam retries a few times anyways if the marker detection fails.)

I don't think the workflow can be improved by freezing the image at any point, but please feel free to write the code for this if you consider it important. If you need help, don't hesitate to ask me when I'm around in the lab.

sedrubal, 2016-05-09:
I think, displaying an outdated image in grayacale instead of colored would be the user friendliest solution but I don't have time to implement this and I think you don't have time neither. So I'm ok with the current solution...

closed 2019-11-01 by mgmax