t-oster/VisiCut #293: Visicut cannot import DXF with a single line

feklee, 2015-08-09:
Sometimes I need to cut something with just a single line, but Visicut throws an error when trying to load my DXF containing just a single line. I've exported it from Rhino, as *R12 Natural*.

My file is named `temp.dxf`. Weirdly there is also an error message about SVG:

[image]

mgmax, 2015-08-10:
could you upload the DXF file somewhere? did you do it via drag-and-drop (which doesn't work as far as I remember)?

feklee, 2015-08-10:
I used *File / Open…*.

`temp.dxf` Base64 encoded:

    ICAwDQpTRUNUSU9ODQogIDINCkhFQURFUg0KICA5DQokQUNBRFZFUg0KICAxDQpBQzEwMDkNCiAg
    OQ0KJERXR0NPREVQQUdFDQogIDMNCmFuc2lfMTI1Mg0KICA5DQokSU5TQkFTRQ0KIDEwDQowLjAN
    CiAyMA0KMC4wDQogMzANCjAuMA0KICA5DQokRVhUTUlODQogMTANCjAuMA0KIDIwDQowLjANCiAz
    MA0KMC4wDQogIDkNCiRFWFRNQVgNCiAxMA0KMTAwLjANCiAyMA0KMC4wDQogMzANCjAuMA0KICA5
    DQokTElNTUlODQogMTANCjAuMA0KIDIwDQowLjANCiAgOQ0KJExJTU1BWA0KIDEwDQoxMi4wDQog
    MjANCjkuMA0KICA5DQokT1JUSE9NT0RFDQogNzANCiAgICAgMA0KICA5DQokUkVHRU5NT0RFDQog
    NzANCiAgICAgMQ0KICA5DQokRklMTE1PREUNCiA3MA0KICAgICAxDQogIDkNCiRRVEVYVE1PREUN
    CiA3MA0KICAgICAwDQogIDkNCiRNSVJSVEVYVA0KIDcwDQogICAgIDENCiAgOQ0KJERSQUdNT0RF
    DQogNzANCiAgICAgMg0KICA5DQokTFRTQ0FMRQ0KIDQwDQoxLjANCiAgOQ0KJE9TTU9ERQ0KIDcw
    DQogICAgMzcNCiAgOQ0KJEFUVE1PREUNCiA3MA0KICAgICAxDQogIDkNCiRURVhUU0laRQ0KIDQw
    DQowLjINCiAgOQ0KJFRSQUNFV0lEDQogNDANCjAuMDUNCiAgOQ0KJFRFWFRTVFlMRQ0KICA3DQpT
    VEFOREFSRA0KICA5DQokQ0xBWUVSDQogIDgNClBMQVRFUyRURU1QJENVVA0KICA5DQokQ0VMVFlQ
    RQ0KICA2DQpCWUxBWUVSDQogIDkNCiRDRUNPTE9SDQogNjINCiAgIDI1Ng0KICA5DQokRElNU0NB
    TEUNCiA0MA0KMS4wDQogIDkNCiRESU1BU1oNCiA0MA0KMC4xOA0KICA5DQokRElNRVhPDQogNDAN
    CjAuMDYyNQ0KICA5DQokRElNRExJDQogNDANCjAuMzgNCiAgOQ0KJERJTVJORA0KIDQwDQowLjAN
    CiAgOQ0KJERJTURMRQ0KIDQwDQowLjANCiAgOQ0KJERJTUVYRQ0KIDQwDQowLjE4DQogIDkNCiRE
    SU1UUA0KIDQwDQowLjANCiAgOQ0KJERJTVRNDQogNDANCjAuMA0KICA5DQokRElNVFhUDQogNDAN
    CjAuMTgNCiAgOQ0KJERJTUNFTg0KIDQwDQowLjA5DQogIDkNCiRESU1UU1oNCiA0MA0KMC4wDQog
    IDkNCiRESU1UT0wNCiA3MA0KICAgICAwDQogIDkNCiRESU1MSU0NCiA3MA0KICAgICAwDQogIDkN
    CiRESU1USUgNCiA3MA0KICAgICAxDQogIDkNCiRESU1UT0gNCiA3MA0KICAgICAxDQogIDkNCiRE
    SU1TRTENCiA3MA0KICAgICAwDQogIDkNCiRESU1TRTINCiA3MA0KICAgICAwDQogIDkNCiRESU1U
    QUQNCiA3MA0KICAgICAwDQogIDkNCiRESU1aSU4NCiA3MA0KICAgICAwDQogIDkNCiRESU1CTEsN
    CiAgMQ0KDQogIDkNCiRESU1BU08NCiA3MA0KICAgICAxDQogIDkNCiRESU1TSE8NCiA3MA0KICAg
    ICAxDQogIDkNCiRESU1QT1NUDQogIDENCg0KICA5DQokRElNQVBPU1QNCiAgMQ0KDQogIDkNCiRE
    SU1BTFQNCiA3MA0KICAgICAwDQogIDkNCiRESU1BTFREDQogNzANCiAgICAgMg0KICA5DQokRElN
    QUxURg0KIDQwDQoyNS40DQogIDkNCiRESU1MRkFDDQogNDANCjEuMA0KICA5DQokRElNVE9GTA0K
    IDcwDQogICAgIDANCiAgOQ0KJERJTVRWUA0KIDQwDQowLjANCiAgOQ0KJERJTVRJWA0KIDcwDQog
    ICAgIDANCiAgOQ0KJERJTVNPWEQNCiA3MA0KICAgICAwDQogIDkNCiRESU1TQUgNCiA3MA0KICAg
    ICAwDQogIDkNCiRESU1CTEsxDQogIDENCg0KICA5DQokRElNQkxLMg0KICAxDQoNCiAgOQ0KJERJ
    TVNUWUxFDQogIDINClNUQU5EQVJEDQogIDkNCiRESU1DTFJEDQogNzANCiAgICAgMA0KICA5DQok
    RElNQ0xSRQ0KIDcwDQogICAgIDANCiAgOQ0KJERJTUNMUlQNCiA3MA0KICAgICAwDQogIDkNCiRE
    SU1URkFDDQogNDANCjEuMA0KICA5DQokRElNR0FQDQogNDANCjAuMDkNCiAgOQ0KJExVTklUUw0K
    IDcwDQogICAgIDINCiAgOQ0KJExVUFJFQw0KIDcwDQogICAgIDMNCiAgOQ0KJFNLRVRDSElOQw0K
    IDQwDQowLjENCiAgOQ0KJEZJTExFVFJBRA0KIDQwDQowLjUNCiAgOQ0KJEFVTklUUw0KIDcwDQog
    ICAgIDANCiAgOQ0KJEFVUFJFQw0KIDcwDQogICAgIDANCiAgOQ0KJE1FTlUNCiAgMQ0KLg0KICA5
    DQokRUxFVkFUSU9ODQogNDANCjAuMA0KICA5DQokUEVMRVZBVElPTg0KIDQwDQowLjANCiAgOQ0K
    JFRISUNLTkVTUw0KIDQwDQowLjANCiAgOQ0KJExJTUNIRUNLDQogNzANCiAgICAgMA0KICA5DQok
    Q0hBTUZFUkENCiA0MA0KMC4wDQogIDkNCiRDSEFNRkVSQg0KIDQwDQowLjANCiAgOQ0KJFNLUE9M
    WQ0KIDcwDQogICAgIDANCiAgOQ0KJFREQ1JFQVRFDQogNDANCjI0NTcyNDUuNDIyODUyOTYzDQog
    IDkNCiRURFVQREFURQ0KIDQwDQoyNDU3MjQ1LjQyMjg2NjczNg0KICA5DQokVERJTkRXRw0KIDQw
    DQowLjAwMDAwMDAxMTYNCiAgOQ0KJFREVVNSVElNRVINCiA0MA0KMC4wMDAwMDAwMTE2DQogIDkN
    CiRVU1JUSU1FUg0KIDcwDQogICAgIDENCiAgOQ0KJEFOR0JBU0UNCiA1MA0KMC4wDQogIDkNCiRB
    TkdESVINCiA3MA0KICAgICAwDQogIDkNCiRQRE1PREUNCiA3MA0KICAgICAwDQogIDkNCiRQRFNJ
    WkUNCiA0MA0KMC4wDQogIDkNCiRQTElORVdJRA0KIDQwDQowLjANCiAgOQ0KJENPT1JEUw0KIDcw
    DQogICAgIDENCiAgOQ0KJFNQTEZSQU1FDQogNzANCiAgICAgMA0KICA5DQokU1BMSU5FVFlQRQ0K
    IDcwDQogICAgIDYNCiAgOQ0KJFNQTElORVNFR1MNCiA3MA0KICAgICA4DQogIDkNCiRBVFRESUEN
    CiA3MA0KICAgICAwDQogIDkNCiRBVFRSRVENCiA3MA0KICAgICAxDQogIDkNCiRIQU5ETElORw0K
    IDcwDQogICAgIDENCiAgOQ0KJEhBTkRTRUVEDQogIDUNCjc3DQogIDkNCiRTVVJGVEFCMQ0KIDcw
    DQogICAgIDYNCiAgOQ0KJFNVUkZUQUIyDQogNzANCiAgICAgNg0KICA5DQokU1VSRlRZUEUNCiA3
    MA0KICAgICA2DQogIDkNCiRTVVJGVQ0KIDcwDQogICAgIDYNCiAgOQ0KJFNVUkZWDQogNzANCiAg
    ICAgNg0KICA5DQokVUNTTkFNRQ0KICAyDQoNCiAgOQ0KJFVDU09SRw0KIDEwDQowLjANCiAyMA0K
    MC4wDQogMzANCjAuMA0KICA5DQokVUNTWERJUg0KIDEwDQoxLjANCiAyMA0KMC4wDQogMzANCjAu
    MA0KICA5DQokVUNTWURJUg0KIDEwDQowLjANCiAyMA0KMS4wDQogMzANCjAuMA0KICA5DQokUFVD
    U05BTUUNCiAgMg0KDQogIDkNCiRQVUNTT1JHDQogMTANCjAuMA0KIDIwDQowLjANCiAzMA0KMC4w
    DQogIDkNCiRQVUNTWERJUg0KIDEwDQoxLjANCiAyMA0KMC4wDQogMzANCjAuMA0KICA5DQokUFVD
    U1lESVINCiAxMA0KMC4wDQogMjANCjEuMA0KIDMwDQowLjANCiAgOQ0KJFVTRVJJMQ0KIDcwDQog
    ICAgIDANCiAgOQ0KJFVTRVJJMg0KIDcwDQogICAgIDANCiAgOQ0KJFVTRVJJMw0KIDcwDQogICAg
    IDANCiAgOQ0KJFVTRVJJNA0KIDcwDQogICAgIDANCiAgOQ0KJFVTRVJJNQ0KIDcwDQogICAgIDAN
    CiAgOQ0KJFVTRVJSMQ0KIDQwDQowLjANCiAgOQ0KJFVTRVJSMg0KIDQwDQowLjANCiAgOQ0KJFVT
    RVJSMw0KIDQwDQowLjANCiAgOQ0KJFVTRVJSNA0KIDQwDQowLjANCiAgOQ0KJFVTRVJSNQ0KIDQw
    DQowLjANCiAgOQ0KJFdPUkxEVklFVw0KIDcwDQogICAgIDENCiAgOQ0KJFNIQURFREdFDQogNzAN
    CiAgICAgMw0KICA5DQokU0hBREVESUYNCiA3MA0KICAgIDcwDQogIDkNCiRUSUxFTU9ERQ0KIDcw
    DQogICAgIDENCiAgOQ0KJE1BWEFDVFZQDQogNzANCiAgICA2NA0KICA5DQokUExJTUNIRUNLDQog
    NzANCiAgICAgMA0KICA5DQokUEVYVE1JTg0KIDEwDQoxLjAwMDAwMDAwMDAwMDAwMEUrMjANCiAy
    MA0KMS4wMDAwMDAwMDAwMDAwMDBFKzIwDQogMzANCjEuMDAwMDAwMDAwMDAwMDAwRSsyMA0KICA5
    DQokUEVYVE1BWA0KIDEwDQotMS4wMDAwMDAwMDAwMDAwMDBFKzIwDQogMjANCi0xLjAwMDAwMDAw
[... 18 lines cut ...]
[... cut ...]
    Mw0KJE1PREVMX1NQQUNFDQogIDENCg0KICAwDQpFTkRCTEsNCiAgNQ0KMjENCiAgOA0KMA0KICAw
    DQpCTE9DSw0KIDY3DQogICAgIDENCiAgOA0KMA0KICAyDQokUEFQRVJfU1BBQ0UNCiA3MA0KICAg
    ICAwDQogMTANCjAuMA0KIDIwDQowLjANCiAzMA0KMC4wDQogIDMNCiRQQVBFUl9TUEFDRQ0KICAx
    DQoNCiAgMA0KRU5EQkxLDQogIDUNCjFEDQogNjcNCiAgICAgMQ0KICA4DQowDQogIDANCkVORFNF
    Qw0KICAwDQpTRUNUSU9ODQogIDINCkVOVElUSUVTDQogIDANCkxJTkUNCiAgNQ0KNkYNCiAgOA0K
    UExBVEVTJFRFTVAkQ1VUDQogMTANCjAuMA0KIDIwDQowLjANCiAzMA0KMC4wDQogMTENCjEwMC4w
    DQogMjENCjAuMA0KIDMxDQowLjANCiAgMA0KRU5EU0VDDQogIDANCkVPRg0K