t-oster/VisiCut #295: Send calibration page failed: Error: null

sedrubal, 2015-08-20:
After clicking `Kamera kallibrieren` > profile: `cut` > `Send Calibration page`, I get this error message box:

```
Error sending page:
null
```

Deleting all settings and downloading them from "FAU FabLab (Erlangen)" does not help. On an other user on the same machine it works...

t-oster, 2015-08-21:
could you start VisiCut from command line and see, which exception is thrown?

sedrubal, 2015-08-26:
Sorry, I can't reproduce it anymore. I think there was no error message in stdout.

t-oster, 2015-08-26:
So if you cannot reproduce it, we can consider this as fixed?

sedrubal, 2015-08-26:
yes, otherwise I'll reopen it ;)

closed 2015-08-26 by sedrubal

Kr4cker, 2018-01-23:
I have the same problem. My VisiCut Version is 1.8-6-gdf69da5f and it's running on Linux Mint 18.3.
If I start the program from the terminal, I get the follow warnings:
`WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.converters.collections.TreeMapConverter (file:/usr/share/visicut/lib/xstream-1.4.3.jar) to field java.util.TreeMap.comparator
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.converters.collections.TreeMapConverter
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release` 

It would be great if somebody can help me with this bug. I would like to use VisiCut to make it easier to laser for our guests.

mgmax, 2018-01-24:
@Kr4cker You may have a look at the changes I proposed in #434 (if you know how to build VisiCut yourself), so that we can get a better error message and find the actual cause.

t-oster, 2018-01-25:
Please try the latest version, it should output the error message more preciseley (thanks @mgmax )

Kr4cker, 2018-01-25:
Alright. Thank you guys. I will test the new version at the weekend and let you know which error message I will get.

Kr4cker, 2018-01-27:
Hey guys, here my new error:
`NullPointerException
com.t_oster.visicut.gui.CamCalibrationDialog.sendButtonActionPerformed(CamCalibrationDialog.java:432)
com.t_oster.visicut.gui.CamCalibrationDialog.access$200(CamCalibrationDialog.java:59)`

I couldn't copy the error. So maybe there are some typos. I checked it twice.
Thanks again for your help =)

t-oster, 2018-01-27:
Seems as If you have Not defined and selected a laserprofile for marking yet

Kr4cker, 2018-01-27:
okay, how can I select a laserprofile? Can you explain me, why there are a laserprofile for each "type of lasering"?

mgmax, 2018-01-27:
@Kr4cker Most of your question should be answered in the documentation: https://github.com/t-oster/VisiCut/wiki/VisiCut-manual . If the documentation is not enough, you are highly encouraged to point out what isn't explained there and extend the documentation with the answers you find out.

Kr4cker, 2018-01-27:
Finally everything works =)
I found a manual how to calibrate the camera. I would like to extend the documentation but I don't know how exactly I can do it. I just started to learn git, but I know how to use markdown ;)
So, how can I help you with the documentation?

t-oster, 2018-01-27:
You can edit the pages right in the browser, no need for knowing git imho

mgmax, 2018-02-17:
@Kr4cker If this bug is now fixed, please mark the github issue as closed. Thanks.

mgmax, 2018-02-17:
@Kr4cker Please ignore my previous message. It was nonsense.

commit 60d9625f (2018-01-27): fix cam calibration "error null" (#435)

* Camera calibration: Error message if no settings are available

Fixes #295

also fixes german localization for the calibration dialog

* Error dialogs: Also show stacktrace for plain Exception("") with empty message