t-oster/VisiCut #303: Make it into the App-Store

t-oster, 2015-09-12:
Instead of just fixing the osx packaging (#278) we should try to add VisiCut to the App Store.

So we are looking for an Apple Developer, who has a license, some free time and can package VisiCut in a way is can be submitted.

hpux735, 2015-09-12:
I’m willing to look into it.  I’m far from an expert, but I’m in the program, and I’ve have something in the store.  My only concern is that there are lots of ways to be rejected.  Are you willing to work with me on code changes that may be necessary?  For example, the sandbox is required these days, so you’re limited with where and how you can access the filesystem.  I had written user preferences into a file in the application bundle in my app, but that was rejected, for example.  In general, I’m for having it in the app store, I think that will really help with adoption and user experience, but it’s going to make packaging harder in the near term, not easier.

- Will

t-oster, 2015-09-12:
Hi, yes. I would like to assist in coding changes, which are necessary. 
Even if packaging is harder, I think if we are once through the process, the rest is only exchanging newer versions of the jar files.
However if it turns out to be too complicated, an old version in the app-store is better than no version. People still can download a newer version manually.

I think the first steps are:
- integrating the JRE into the app-distribution
- either exchanging JavaApplicationStub with a correct version, or switch to a custom launcher written in ObjectiveC/Cocoa
- Make code changes necessary for integration in the app store
- sign the package
- find a way to automate this process for newer versions

hpux735, 2015-09-13:
I’ve started working on it, and it seems to be going well enough so far.  I’m in the middle of figuring out how to sign mkbitmap and potrace correctly.  I have a build of the app that is signed well enough to distributed that shouldn’t make the user’s computer complain, but it’s not good enough for the mac store yet.

I’ll let you know how it goes.  I suspect I’ll be able to get it to the point where I can submit it without any changes to the java code, but that will still probably be rejected.

t-oster, 2015-09-13:
Hi, I think before the signing and submitting, fixing the app-bundle, so that it runs on all OSX version (with and without java installed) would be an important step. It would fix many issues ( #183 #210 #146 #278 ). So if you have an app bundle with a working JavaApplicationStub and JRE included, please create a pull-request, so it is used for the official builds. Right now I have to tell all OSX users to use the platform-independent zip version.

hpux735, 2015-09-13:
The signing part is actually the easiest thing to do.  Also, submitting (though not requesting review) allows me to send beta testing links that people can use to try it out and provide feedback.

I have to confess that I have basically no java experience, and have no idea how to include the JRE in the application bundle.  If anyone knows how to do that, and can provide an application bundle that does it, I can get it signed easily enough.

Does anyone know of another application that uses java that is in the app store?  I know that Arduino, Processing, and Energia (through they’re all basically the same thing) all use Java, but also don’t include the JRE.

dreydean, 2015-12-25:
Hey guys! How is this project coming with integration of visicut into the appstore/

hpux735, 2015-12-26:
I’m no good at Python, so I gave up.  But, if someone can get an Xcode project that builds it, I can code sign and upload it to the app store because I have access.

mgmax, 2015-12-26:
I don't own any Apple computer, so I can't really help, but here are some links to follow:
http://www.intransitione.com/blog/take-java-to-app-store/
https://bitbucket.org/infinitekind/appbundler

Java programming knowledge isn't really necessary, for the first start you just have to puzzle together all parts of the app bundling system, see how far you come and upload it (preferably fork the repository here). If you're stuck with one specific problem, someone else will hopefully be able to help.

mgmax, 2015-12-26:
An easier route might be DCP setupmaker https://github.com/DevComPack/setupmaker , which is a frontend for the cross-platform izPack installer creation tool. This could also be a good replacement for the Windows installer.

dreydean, 2015-12-26:
Why Python? Why does it not need to be in Swift?

dreydean, 2015-12-26:
Also is there a way we can just convert it over to JavaScript and then we can build an app with meteor and host it online.

hpux735, 2015-12-26:
I do kinda like the idea of JavaScript. I use octoprint on my 3d printer, and I like that use case. Also, the material and machine presets would stay with the machine, rather than have to be copied for every user.

peteruithoven, 2015-12-26:
@dreydean I think that's the subject of another issue: https://github.com/t-oster/VisiCut/issues/232

mgmax, 2015-12-26:
@dreydean There is no way of "just" converting a program from one language to another, especially when it comes to the user interface. To some extent, converting (cross-compiling) works for back-end code like the drivers. The main work for a web-based VisiCut would be the front-end - the "webpage" you use to upload a file, look at the preview and adjust the settings; and the connection between the frontend and the existing codebase.

This issue here is about how the existing Java VisiCut program can be *packaged*  -- not rewritten -- for submission to the Mac OS X AppStore, so that people can install it more easily on desktop computers running MacOS.

mgmax, 2019-05-28:
Making it into the Mac Appstore may be outright impossible due to legal issues, because GPLed software is forbidden by Apple. I could not find clear sources if the same is true for LGPL or not. I also don't know what that means for bundling Java (OpenJDK is GPL2).

https://en.wikipedia.org/wiki/Mac_App_Store  says: 

> Apps must be approved by Apple before becoming available on the store. Disallowed types of apps revealed by Apple include apps that: ..... are free software licensed only under GPL (because the App Store Terms of Service imposes additional restrictions incompatible with the GPL).

It's disappointing to see how much Apple locks down everything :-(

t-oster, 2019-05-28:
How do the other FOSS applications handle this? Is there nothing fancy like Thunderbird, Firefox, OpenSCAD, Inkscape, LibreOffice in the App Store?

hpux735, 2019-05-28:
This software doesn't have to be GPL.  GPL is not a great license.  It's actually pretty restrictive.  If it were Apache, LGPL, BSD, etc., it would be fine.

hpux735, 2019-05-28:
And, it looks like it's already LGPL.  So, it should not be a problem.

mgmax, 2019-05-28:
I'm just at the Inkscape Hackfest where we decided that it's more or less impossible to get into the App Store because Apple disallows GPL code.
VLC did lots of relicensing before they could use the App Store.

@hpux735 As I noted, OpenJDK is GPL, and we need to bundle it because VisiCut is built on Java. The only alternative would be paying for a commercial JDK license. 
The problem is not with the GPL license, which can be used without problems e.g. on Windows Store and Steam Store, but with Apple's excessively restrictive guidelines.

t-oster, 2019-05-28:
So there is no other open source Java app in the app store?

hpux735, 2019-05-28:
I  have a hard time believing that there are no other GPL or Java apps in the app store.  It's probably worthwhile contacting apple rather than relying on Wikipedia.