t-oster/VisiCut #304: Limit selectable resolutions to list?

peteruithoven, 2015-09-20:
Currently the drivers seem to determine which resolutions (dpi) are available, probably the ones presented in the resolution dropdown. I might have forgot the reason, but why are we still allowing other numbers into that field? Why not limit it to that list?

t-oster, 2015-09-21:
The problem is: Some Lasercutters (like the Epilog ones) only allow a fixed set of resolutions, which are used to map the units like cm to stepper-motor-steps. The jobs are then sent in full "steps" together with the resolution. Other drivers, like LAOS and Smoothieboard do this conversion on their own and get the jobs in "mm" without any resoltuion. In VisiCut, the resolution is still used for raster conversion and for converting round shapes to small straight lines before sending the job. But a laos lasercutter basically accepts any resolution, which is the reason why you can specify a resolution, which is not listed in the driver, but may be e.g. the natural fit for your device (depending on the real stepper-motors and your setup).

peteruithoven, 2015-09-22:
But when I use a LaOS lasercutter and a unsupported resolution, like 250 I get the following error when sending a job to the lasercutter. 
> The Lasercutter you selected, does not support 250.0dpi. 
> Do you want to use 200dpi instead? 

(I can remember we improved this error message at one time)