t-oster/VisiCut #308: calculating preview takes ages, even on cut-only files

mgmax, 2015-10-02:
Although I can't clearly reproduce this bug, I am still reporting it because I have heard it from lots of different users:
Sometimes VisiCut takes many minutes calculating the engrave preview and also calculating the laser job, although the file has no engrave part. This happens in mixed mappings (like FAU FabLab Erlangen default: map-by-color, cut red, ..., engrave rest) and disappears as soon as the engrave part is disabled in the mapping.

These files take much longer to calculate than an engrave calculation would take otherwise.

I'm sorry that I can't give a better bugreport, but maybe someone else can contribute a file that always shows the error.

mgmax, 2015-10-02:
- open VisiCut (FAU FabLab settings)
- open or drag-and-drop file: https://gist.githubusercontent.com/mgmax/dac2e8de0dba28c61d56/raw/d345a2b4fb2445cda98b603a6e7f90fdc06de269/long-calculate.svg
- click in the preview area, but outside of the object, to deselect it. Then the preview will be calculated, although there is nothing to engrave. It will not be recalculated on resizing however.
- press the 'calculate time' button, it takes quite long and shows a result of ~6min (for a certain material)
- move the object, then right-click and 'reset position'
- calculate time again, now it says 0:54min. strange.

On Windows, the behaviour is slightly different, once I tried it it was still calculating after 15min.

There is a difference between deleting the file from VisiCut and then re-opening it versus just right-click and 'refresh'. Having a fresh VisiCut start also seems to cause the error more often, although I'm not very sure about that.

thinkl33t, 2015-11-12:
I found this was happening when i tried the TSP optimiser.  Switched back to the regular one and it worked again.

mgmax, 2015-11-12:
what I describe happens with the inside-out sorting. That the TSP optimiser is only academic and not really usable in practice is another problem.