t-oster/VisiCut #310: Marlin driver - connects (can see response back) but GUI hangs on "Opening COM36"

openhardwarecoza, 2015-11-07:
Hi

I have a Marlin based laser (http://openbuilds.org/builds/openbuilds-freeburn-2-mini-v-slot-co2-laser-40-50w.1657/) hooked to my machine.  no issues communicating with it using Pronterface. 

I tried the new Marlin driver using the precompiled binaries from https://drive.google.com/file/d/0Byu4961_yqM9SWppVEVIYkltNlE/view?pageId=113003636260926549707

If I run the jar I get output in CMD from the headers of the Marlin startup - showing me the serial is connecting, however the GUI gets stuck on 'opening 'COM36'
Com 36 is the correct port, and baud rate is correct too (115200) 
[image]

In pronterface I know the connect function waits for the string "start' from the firmware before it shows connected, not sure how libLasercut handles that?  I assume its something like that since it gets stuck on opening port and doesnt move on to start sending my startup gcode (homing, etc) and then the job...

openhardwarecoza, 2015-11-07:
Suspect I found my answer after mentioning how pronterface waits for 'start'

openhardwarecoza, 2015-11-07:
Nope, enabled SD support in Marlin, and only got three lines  further before it doesnt do anything more:

[image]

@quillford, any ideas?

quillford, 2015-11-07:
Do you have a link to the Marlin fork you use? I wrote the driver based on Turnkey's Marlin fork.

openhardwarecoza, 2015-11-08:
I am using the Lansing fork https://github.com/lansing-makers-network/buildlog-lasercutter-marlin - but will try Turnkey's fork too...
But I don't have a SD Card hooked up (my controller: http://openbuilds.org/threads/openbuilds-freeburn-2-mini-v-slot-co2-laser-40-50w.1112/page-4#post-16019)

Since yesterday I've setup Netbeans and got dug into the code. 
I cloned the latest visicut dev branch, and then replaced the /lib/LibLaserCut with the LibLasercut-smoothieboard fork where your driver is included (its not included in the development branch? @t-oster 

In drivers/Marlin.java

I changed

   while(!(waitForLine().startsWith("echo:SD")))

    to

   while(!(waitForLine().startsWith("echo:SD init fail")))

Built LibLaserCut then built Visicut, but still same behaviour

Here's the dump my marlin build dumps on connection

rinter is now online.
echo: External Reset
Marlin 1.0.0
echo: Last Updated: Nov  7 2015 14:19:05 | Author: (Peter vd Walt)
Compiled: Nov  7 2015
echo: Free Memory: 2540  PlannerBufferBytes: 2576
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X53.50 Y53.50 Z3260.00 E500.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z100 E10000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B1 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Laser lifetime usage:
echo: Hours: 0
 Minutes: 0
echo:PID settings:
echo:   M301 P22.20 I1.08 D114.00
echo:SD init fail

Let me go try Turnkey ... I'll be back to report in a few mins

openhardwarecoza, 2015-11-08:
OK, so with the fork from:  https://github.com/TurnkeyTyranny/buildlog-lasercutter-marlin

my output on startup is  (raw as downloaded, no changes even to configuration.h)

start
Printer is now online.
echo: External Reset
Marlin 1.0.0
echo: Last Updated: Nov  8 2015 06:42:53 | Author: (none, default config)
Compiled: Nov  8 2015
echo: Free Memory: 2272  PlannerBufferBytes: 2576
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X157.48 Y157.48 Z6047.24 E7600.00
echo:Maximum feedrates (mm/s):
echo:  M203 X7600.00 Y7600.00 Z10.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X2600 Y2600 Z2 E2
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S2000.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Laser lifetime usage:
echo: Hours: 0
 Minutes: 0
echo:PID settings:
echo:   M301 P22.20 I1.08 D114.00
echo:SD init fail

Still pretty much the same as the lansing fork.

quillford, 2015-11-08:
As long as the last line of Marlin's initial dump starts along the lines of "echo:SD" it should be able to identify it as using the correct driver. Note: when modify the driver, you have to create a new profile using that driver or else it won't use your changes (however the change you made would make no difference). Does VisiCut display any error message or does it just hang? It is worth noting I've only tested this driver on OS X and Linux, and it works fine on both.

openhardwarecoza, 2015-11-08:
Aha might be due to 'need to create new profile' - been using the same profile each time

Lets give that a go

openhardwarecoza, 2015-11-08:
Success!

With three changes though.

Line 43 of Marlin.java 
    setInitDelay(5);

If this is set to 0, even with a new profile, still no go.

    while(!(waitForLine().startsWith("echo:SD")))

Checking startswith "echo:SD" covers both successful and failed SD init.

and lastly, adding a new profile made it work!

openhardwarecoza, 2015-11-08:
Ok initial hurdle crossed, but it sends the last command in me pre-gcode, and then nothing more (status in visicut still displays 'sending' but nothing happening...

Suspect it might be the dev branch + smoothie liblasercut.  Going to try with the smoothie branch entirely

openhardwarecoza, 2015-11-08:
Cloned the smoothie branch, which recusrively cloned liblasercut from the smoothieboard branch automatically.  
Still same issue though, dxf and svg.  Sends the start gcodes and then nothing more - kind of like it didnt generate gcode from the svg to send?

t-oster, 2015-11-08:
Does it send post-job-code?maybe try the right-click "move to position" feature,could it be thst you did not select a mapping for your file?

openhardwarecoza, 2015-11-08:
The plot thickens!

a) I dont have a right click, move to position menu
[image]

b) I havent seen the mappings thing in my playing, so hunted quickly through the menus: 
[image]

Wait what? I dont have a mappings tab in my main window...

openhardwarecoza, 2015-11-08:
Sorry noob error:  Move to pos is available if i first click somewhere else in the viewer, then back on it :/

So heres a screencap: 

1.  Loaded file, mapped, executed.  Sent my pre-gcode (G21) 
2.  Sat on 'sending' nothing happens
3.  Tried a move to position, got port in use exception! (still in use from the Execute? 

[image]

quillford, 2015-11-08:
Can you scroll down in that terminal window and post a screenshot of that? It shows that gcode was sent.

closed 2015-11-11 by openhardwarecoza