t-oster/VisiCut #312: Smoothieboard and Visicut ver1.7-290 Circle Cutting Issue

xboxhacker, 2015-11-16:
Using a Smoothie board and Visicut ver 1.7-290 on XP SP3.

Loaded the example file "circle-20mm-red.svg"

Setup to cut all 35% power and 300mm movement.

When the machine goes to cut the circle; it only cuts one segment, then stops (laser off), and then cuts another...etc. It does not cut the circle in one continuous motion.

If I use Inkscape and a Laser gcode exporter, to make a 20mm circle. Load it to pronterface. It cuts the circle in one pass, no stopping.

Please help.

Thank you!

t-oster, 2015-11-17:
Are you using serial or Ethernet?

xboxhacker, 2015-11-18:
I am using Ethernet. I have also tried a Win7 laptop. With the same results.

I will try USB in the morning, and report back.

Thanks!

xboxhacker, 2015-11-18:
OK. Couldn't wait til the AM.

I cannot connect to the laser at all via USB. I tried changing the auto to 5 (my com). Still no joy. But I can via prontwrface via USB on Com5.

For the record, I am using a MKS SBase board.

t-oster, 2015-11-18:
I think it has to be "COM5" on windows, not just "5"....

xboxhacker, 2015-11-18:
OK, here is what I have tried. Still no USB connection.
-5
-com5
-Com5
-COM5
-auto
-AUTO
-115200 (baud speed)
-increased connection time out

All the while i have been able to connect via USB on pronterface.

Is there a way to view/debug the gcode that is being sent to the laser?

Straight line cuts are fine, just arcs and circles are a problem.

t-oster, 2015-11-19:
Hi, sorry for the long delay, but I am quite busy (as usual). Could you try the HTTP-Upload feature (you have to start the job via smoothieboards web interface afterwards)? I suspect the problem being, that the driver always waits for "OK" after each command. 
You can see the gcode when starting VisiCut from the command line, by the way. The code is located https://github.com/t-oster/LibLaserCut/blob/smoothieboard/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java here, if you want to understand the underlying stuff.
As you can see here https://github.com/t-oster/LibLaserCut/blob/smoothieboard/src/com/t_oster/liblasercut/drivers/SmoothieBoard.java the Smoothieboard driver is just the GenericGcodeDriver with some settings set to fixed values, so you may try to use the GenericGcodeDriver with the same values instead, so you can i.e. disable the SETTING_WAIT_FOR_OK setting and try without.
I think it should work with small jobs, but someone reported, that if you cut very large jobs, the buffer will overflow... question is if this is the case on ethernet, too.

t-oster, 2015-12-02:
Hi, I just implemented the auto-play feature for HTTP Upload. So please do the following:
* Get the latest VisiCut
* Connect your Smoothieboard via Ethernet
* In the driver's settings make sure that:
 * IP/Hostname is empty
 * HTTP Upload URL is http://[your-smoothie-ip]/upload
 * COM Port is empty
 * Start Job after HTTP Upload is checked

Now your jobs should be uploaded to smoothieboard and started if you press execute in VisiCut. You can re-play old jobs on smoothieboards webinterface (http://[your-smoothie-ip]/index.html)

Please report if the problem persists. If not, the "wait-for-ok after each line" may be the culprit.

DavidJRichards, 2015-12-07:
Greetings Thomas, I have the same problem with delay between cutting sections of a circle.
I'm using an Ethernet connection to a current version of smoothiware (Nov 25 2015), If I save the file, upload using Pronterface, and then cut the sections are cut without gaps.

I compiled and ran the latest version on a new 64 bit Linux mint installation to do the test, I cant get a com port working to try with on my board - only etehrenet and USB at present. It looks as if the wait for ok is the problem, I'll try again with this un-selected to be sure.
I have tried to use the upload and cut method you describe above but the upload never completes. 

I cant understand what the problem could be trying to upload from within visicut.
Kind regards, David.

t-oster, 2015-12-16:
Could you test if when using a huge file without the "wait-for-ok", if there is an overflow after some time?
I think we get problems if we remove it completely.
For ehternet connection the http-upload should solve the problem and is the way to go in my opinion, but for serial/usb I think a possiblity is to wait for ok, but alwas send i.e. 10 commands in advance and then wait for the first to have completed before sending the 11th.

DavidJRichards, 2015-12-16:
Hello, I shall not be able to test now for a while, i'm sure you are right and some kind of flow control will be needed for serial work. flow control would not introduce the delay seen by waiting for an ok response. I see by studying the ethernet transfers with wireshark that several 'ok' responses are sent together in a single frame, odd but working well enough. I'm happy to use network data, its a pity the httpnfile upload feature doesnt work from within visicut though. David

t-oster, 2015-12-17:
Well, I guess that may be the culprit: As soon as the command is executed, the smoothieboard sends the OK, but it does not flush the buffer, so the OK gets buffered until the buffer is full or a timeout occurs and then it is sent. So maybe a buffer-flush on the smoothieboard side would be the cleanest solution.

If we do not wait for ok, we risk an overflow on the smoothieboard if too much commands are sent at once (imagine big engraving jobs).

The upload feature works on my board, so we should look further into that issue why it does not work on yours.

Serial (if it works) may have the same buffer problem.

fiskfan, 2016-01-02:
Hi,

I think i experiance similar issues as described here. I have tried Ethernet and COM, the same issue. Until reading this, I was not aware that the HTTP upload feature did post/create a file which then is accessible thru the built in web interface of the smoothieboard.

T-Oster, i did film the issue and as you can see in this video it moves fast for a short wile and then it stops, moves, stops, moves. This job is executed via ethernet: https://www.youtube.com/watch?v=PLRf07kquHU

I will try to run the job thru the web interface as you suggested earlier to see if it runs smoothly.

xboxhacker, 2016-01-03:
Hello, sorry for my response delay.

I have downloaded the new version 1.7-309. When i hit execute, it says sending. Then hangs. It does nothing at this point.

I will edit the WAIT FOR OK and see what happens.

Thanks!

xboxhacker, 2016-01-04:
OK, Changed the "WAIT FOR OK" to false. It cut the 20mm circle fine.

So I then loaded up a very large file. Fully detailed Millennium Falcon SVG. It started, then finished 5 seconds after start. Cutting about 0.5% of the total file. So it will not work with larger files.

Still cannot get Visicut to work with HTTP Upload. But I am able to upload to the SD via the browser web app. So I know uploading is not an issue.

t-oster, 2016-01-04:
Hi please send a screen shot of the laser cutter config. 
-- 

Diese Nachricht wurde von meinem Android-Mobiltelefon mit K-9 Mail gesendet.

fiskfan, 2016-01-04:
Hi,
I tested the upload feature and is stuck in "Sending" and the file never shows up on the smoothieboard web page. My smoothieboard version is edge-43b718c, Build Date Dec 13 2015. 

I have attached a couple of screenshots showing the config im currently running. I have tried to change the "Wait MS" from 1000 to 10 but it did not make any difference now when im testing the HTTP upload feature.

Any ideas?

[image]
[image]
[image]
[image]
[image]

t-oster, 2016-01-04:
Please open the smoothieboard web-interface in chrome or firefox, press F-12 for the developer tools and upload a file manually to see what happens in the network tab. Maybe they changed the URL or something

fiskfan, 2016-01-04:
Hi,
It looks like "upload" is still used?
[image]

t-oster, 2016-01-04:
please click on that entry to view the request headers

t-oster, 2016-01-04:
the code is here https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java#L428
Do you have NetBeans etc installed so that you can debug?

fiskfan, 2016-01-04:
Okey. Updated screenshot showing response headers.
I had a look in the code and the upload part does not look that complicated. However, i have very little knowledgment about software dev and have never worked with NetBeans for debugging...

[image]

xboxhacker, 2016-01-04:
Deeper in to the rabbit hole......

I debuged in Netbeans. The http upload and run works 100%!!!!

Went back to the Windows version, it does not work....

Tried to compile Visicut, so I could just use that one. I get this error:

-------------run.classpath------- lib/kabeja-0.4/lib/batik-all.jar:lib/kabeja-0.4/lib/js.jar:lib/kabeja-0.4/lib/kabeja-0.4.jar:lib/kabeja-0.4/lib/kabeja-svg-0.4.jar:lib/kabeja-0.4/lib/kabeja-xslt.jar:lib/kabeja-0.4/lib/miethxml-toolkit.jar:lib/kabeja-0.4/lib/miethxml-ui.jar:lib/kabeja-0.4/lib/pdf-transcoder.jar:lib/kabeja-0.4/lib/xml-apis-ext.jar:lib/kabeja-0.4/lib/xml-apis.jar:lib/freehep-psviewer-2.0-standalone.jar/:lib/appframework-1.0.3.jar:lib/beansbinding-1.2.1.jar:lib/swing-worker-1.1.jar:lib/gtkjfilechooser.jar:lib/AppleJavaExtensions.jar:lib/commons-net-3.1.jar:lib/LibLaserCut/dist/LibLaserCut.jar:../LibLaserCut-develop/dist/LibLaserCut.jar:C:\Users\billbill\Desktop\VisiCut-develop\lib:C:\Users\billbill\Desktop\VisiCut-develop\lib:lib/svg-salamander-core.jar:lib/xstream-1.4.3.jar:lib/xpp3-1.1.4c.jar:lib/javassist-3.16.1-GA.jar:lib/ognl-3.0.5.jar:lib/slf4j-api-1.7.5.jar:lib/slf4j-nop-1.7.5.jar:lib/thymeleaf-2.0.17-SNAPSHOT.jar:lib/json-simple-1.1.1.jar:lib/scribe-1.3.7_useragent_mod.jar:lib/commons-io-2.4.jar:lib/webcam-capture-0.3.11-20141227.105512-4.jar:lib/bridj-0.6.2.jar:lib/slf4j-api-1.7.2.jar:lib/httpmime-4.3.6.jar:lib/httpclient-4.3.6.jar:lib/httpcore-4.3.3.jar:lib/BrowserLoginDialog.jar:lib/AbsoluteLayout.jar:lib/com.objectplanet.image.PngEncoder.jar:lib/zxing-3.2.1/core-3.2.1-20150223.162544-2.jar:lib/zxing-3.2.1/javase-3.2.1-20150223.162559-2.jar:C:\Users\billbill\Desktop\purejavacomm-master:build/classes
--------- build.classes.dir.resolved ------ C:\Users\billbill\Desktop\VisiCut-develop\build\classes
-------- run.classpath.without.build.classes.dir ---- C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\batik-all.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\js.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\kabeja-0.4.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\kabeja-svg-0.4.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\kabeja-xslt.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\miethxml-toolkit.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\miethxml-ui.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\pdf-transcoder.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\xml-apis-ext.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\kabeja-0.4\lib\xml-apis.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\freehep-psviewer-2.0-standalone.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\appframework-1.0.3.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\beansbinding-1.2.1.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\swing-worker-1.1.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\gtkjfilechooser.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\AppleJavaExtensions.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\commons-net-3.1.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\LibLaserCut\dist\LibLaserCut.jar;C:\Users\billbill\Desktop\LibLaserCut-develop\dist\LibLaserCut.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib;C:\Users\billbill\Desktop\VisiCut-develop\lib\svg-salamander-core.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\xstream-1.4.3.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\xpp3-1.1.4c.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\javassist-3.16.1-GA.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\ognl-3.0.5.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\slf4j-api-1.7.5.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\slf4j-nop-1.7.5.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\thymeleaf-2.0.17-SNAPSHOT.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\json-simple-1.1.1.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\scribe-1.3.7_useragent_mod.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\commons-io-2.4.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\webcam-capture-0.3.11-20141227.105512-4.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\bridj-0.6.2.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\slf4j-api-1.7.2.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\httpmime-4.3.6.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\httpclient-4.3.6.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\httpcore-4.3.3.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\BrowserLoginDialog.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\AbsoluteLayout.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\com.objectplanet.image.PngEncoder.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\zxing-3.2.1\core-3.2.1-20150223.162544-2.jar;C:\Users\billbill\Desktop\VisiCut-develop\lib\zxing-3.2.1\javase-3.2.1-20150223.162559-2.jar;C:\Users\billbill\Desktop\purejavacomm-master;
--- build.classes.dir --- build/classes
C:\Users\billbill\Desktop\VisiCut-develop\build.xml:95: Warning: Could not find resource file "C:\Users\billbill\Desktop\VisiCut-develop\lib\LibLaserCut\dist\LibLaserCut.jar" to copy.
BUILD FAILED (total time: 22 seconds)

t-oster, 2016-01-04:
looks as if you are trying to build VisiCut without building LibLaserCut first. Did you open LibLaserCut as NetBeans Project as described in https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started?

xboxhacker, 2016-01-04:
not gonna lie... I am not a programmer....

Useing Git Bash, I downloaded the repo. Opened LibLaserCut, built, then Visicut. It build fine this time. Now all new set of problems.......

## When i debug from netbeans, it works 100%! Here is the output:

ant -f C:\\Users\\billbill\\VisiCut -Dnb.internal.action.name=debug -Ddebug.class=com.t_oster.visicut.gui.VisicutApp debug
init:
Deleting: C:\Users\billbill\VisiCut\build\built-jar.properties
deps-jar:
Updating property file: C:\Users\billbill\VisiCut\build\built-jar.properties
LibLaserCut.init:
LibLaserCut.deps-jar:
Updating property file: C:\Users\billbill\VisiCut\build\built-jar.properties
LibLaserCut.compile:
Copy libraries to C:\Users\billbill\VisiCut\lib\LibLaserCut\dist\lib.
To run this application from the command line without Ant, try:
java -jar "C:\Users\billbill\VisiCut\lib\LibLaserCut\dist\LibLaserCut.jar"
LibLaserCut.jar:
compile:
> G21
> G90
> M3
> M106
> G0 X21.564600 Y11.252200 F3600
> G1 X21.513800 Y12.268200 S1.000000 F1200
> G1 X21.361400 Y13.258800
> G1 X21.107400 Y14.198600
......
> G1 X21.564600 Y11.252200
> G1 X21.564600 Y11.252200
> G0 X0 Y0
> M5
> M107
> M84
Response: hasError:false,code:200,message:OK,data:
OK

Response: hasError:false,code:200,message:OK,data:
Playing /sd/circle-20mm-red.svg.gcode
  File size 1725

## When I run the JAR from a terminal it DOES NOT work. Here is the output:

C:\Users\billbill>java -jar "C:\Users\billbill\VisiCut\dist\Visicut.jar"
> G21
> G90
> M3
> M106
> G0 X21.564600 Y11.252200 F3600
> G1 X21.513800 Y12.268200 S1.000000 F1200
> G1 X21.361400 Y13.258800
> G1 X21.107400 Y14.198600
......
> G1 X21.513800 Y10.210800
> G1 X21.564600 Y11.252200
> G1 X21.564600 Y11.252200
> G0 X0 Y0
> M5
> M107
> M84
Exception in thread "Thread-6" java.lang.NoClassDefFoundError: net/sf/corn/httpclient/HttpClient
        at com.t_oster.liblasercut.drivers.GenericGcodeDriver.http_upload(GenericGcodeDriver.java:428)
        at com.t_oster.liblasercut.drivers.GenericGcodeDriver.disconnect(GenericGcodeDriver.java:621)
        at com.t_oster.liblasercut.drivers.GenericGcodeDriver.sendJob(GenericGcodeDriver.java:681)
        at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:754)
        at com.t_oster.visicut.gui.MainView$62.run(MainView.java:1982)
Caused by: java.lang.ClassNotFoundException: net.sf.corn.httpclient.HttpClient
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        ... 5 more

xboxhacker, 2016-01-05:
OK, I got it!!!

I went to visicut proterties, in netbeans and added the net/sf/corn/httpclient/HttpClient  .jar file to the list! 

## It is now working 100% :)

Thank you for all your help.

t-oster, 2016-01-05:
Sorry, I must have forgotten to add the library to VisiCut aswell. Should be fixed in the current build. So @fiskfan and @DavidJRichards please try the upload feature again with the current build.

fiskfan, 2016-01-05:
Hi,
http://download.visicut.org/ gives me 502 Bad Gateway ?

t-oster, 2016-01-05:
my httpd keeps locking itself up. I restarted it. Any suggestions for a good platform to upload builds?

fiskfan, 2016-01-05:
Okey. so uploading seems to be working better now. But i'm experiencing veeery slow upload rates.
Im testing with a jpg image scaled to 2x1,5 cm and running a engrave job. 500 DPI seems to be able to upload and the file is around 1,7 MB. 

But if i increase the DPI to 1000 then it seems as if it never completes the upload. If i abort the upload and check the gcode file on the SD card then it seems as if it has stopped in the middle of the transaction at around 1,8 MB. It was uploading for around 10 minutes before i aborted it by closing visicut.

Uploading a large gcode file form the smoothie web runs much faster. - Has anyone else noticed similar behavior with slow uploads?

Does visicut create a gcode file somewhere on the machine its running or does it only create one during upload session?

t-oster, 2016-01-30:
The filename can be configured in the lasercutter-settings 
[image]. Just exchange the space with a hyphen and you're good to go.

t-oster, 2016-01-30:
Please create another issue for the slow transfer, since this issue has already gotten very long.

closed 2016-01-30 by xboxhacker