t-oster/VisiCut #317: Use another open file dialogue

sedrubal, 2015-11-25:
Please use another standard open file dialogue. I think there is no need in creating a own open file dialogue. As far as I know, every GUI framework has a open file dialogue for every supported OS. The current open file dialogue is quite unusable as there are missing many standard features.

t-oster, 2015-11-26:
Hi, it depends on the OSX and Java Version, which dialog is used. I wanted to use the OS X default dialog, and for that I already did a switch in the past, but this could have changed in between.
In Java we do not really have a choice, either it's AWT or Swing. As you can see here https://github.com/t-oster/VisiCut/blob/develop/src/com/t_oster/visicut/gui/MainView.java#L1727 it uses FileDialog on OSX and JFileChooser on everyhting else. Usually JFileChooser is more advanced, but FileDialog at least in the past did use the OSX system dialog. Maybe we should now switch to JDialog? Who can test this on OSX?

t-oster, 2016-03-27:
the switch between FileDialog (AWT) and JFileChooser (Swing) is done here https://github.com/t-oster/VisiCut/blob/develop/src/com/t_oster/visicut/gui/MainView.java#L1895 

Also here https://github.com/t-oster/VisiCut/blob/14b36f14c1fc34fb01c80789563f973064b70ebd/src/com/t_oster/visicut/gui/VisicutApp.java#L163 was a gnome-specific version of jfilechooser which was removed due to a bug.

According to https://code.google.com/archive/p/gtkjfilechooser/ the FileDialog now uses the native version also on gnome, so maybe @z4ziggy is right in #351 and we should make this default with linux also.

We need some KDE and some Windows user to check which one works better.

t-oster, 2016-03-27:
Please download the FileDialog version from here http://download.visicut.org/feature-filedialog and try.
On Cinnamon it uses the native version and thus is much better:

FileDialog@LinuxMint/Cinnamon/OpenJDK7
[image]

JFileChooser@LinuxMint/Cinnamon/OpenJDK7
[image]

mgmax, 2016-03-27:
feature-filedialog @ Debian 8  KDE4 - native file dialog :+1: , works except for any file from "Recently Used" :-1: 
[image]

Trace when using 'Recently Used':
```
     [java] Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
     [java]     at java.io.File.<init>(File.java:277)
     [java]     at com.t_oster.visicut.gui.MainView.openFileDialog(MainView.java:1918)
     [java]     at com.t_oster.visicut.gui.MainView.openMenuItemActionPerformed(MainView.java:1943)
     [java]     at com.t_oster.visicut.gui.MainView.access$2600(MainView.java:131)
     [java]     at com.t_oster.visicut.gui.MainView$32.actionPerformed(MainView.java:1231)
     [java]     at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2018)
     [java]     at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2341)
     [java]     at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
     [java]     at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
     [java]     at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
```

t-oster, 2016-03-29:
Yep. There is a Bug in FileDialog, which prevents to get the correct path of a file, if it was not selected from an existing directory. So the "virtual directories" like "recently used" or "search" produce a null pointer exception.
I could not find a way to get the correct directory for those files.

commit bfecb02b (2016-04-18): Use FileDialog also on Linux. Fixes #317
closed 2016-04-18 by t-oster