t-oster/VisiCut #320: Exception in thread "AWT-EventQueue-0" java.lang.UnsupportedClassVersionError: com/google/zxing/LuminanceSource : Unsupported major.minor version 51.0

peteruithoven, 2015-12-16:
When starting Visicut 1.7-297 (the latest) in OS X 10.9.5 it got stuck at the loading screen. 
(OS X 10.9.5 is two version behind the latest, see [OS X versions](https://en.wikipedia.org/wiki/OS_X#Versions))
Starting it through the terminal (`VisiCut.app/Contents/MacOS/JavaApplicationStub`) it gives the following error: 
```
MacOS/JavaApplicationStub 
Dec 16 17:12:38 peteruithoven-2.local JavaApplicationStub[25828] <Error>: The function `CGContextErase' is obsolete and will be removed in an upcoming update. Unfortunately, this application, or a library it uses, is using this obsolete function, and is thereby contributing to an overall degradation of system performance.
Exception in thread "AWT-EventQueue-0" java.lang.UnsupportedClassVersionError: com/google/zxing/LuminanceSource : Unsupported major.minor version 51.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
	at com.frochr123.qrcodescan.QRCodeScanner.<init>(QRCodeScanner.java:48)
	at com.frochr123.periodictasks.RefreshQRCodesTask.<init>(RefreshQRCodesTask.java:69)
	at com.t_oster.visicut.gui.MainView.setCameraActive(MainView.java:3401)
	at com.t_oster.visicut.gui.MainView.refreshButtonStates(MainView.java:1662)
	at com.t_oster.visicut.gui.MainView.visicutModel1PropertyChange(MainView.java:2129)
	at com.t_oster.visicut.gui.MainView.<init>(MainView.java:382)
	at com.t_oster.visicut.gui.VisicutApp.startup(VisicutApp.java:72)
	at org.jdesktop.application.Application$1.run(Application.java:171)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:209)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:715)
	at java.awt.EventQueue.access$400(EventQueue.java:82)
	at java.awt.EventQueue$2.run(EventQueue.java:676)
	at java.awt.EventQueue$2.run(EventQueue.java:674)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:86)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:685)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:296)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:211)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:196)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:188)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:122)
java.net.SocketTimeoutException: connect timed out
```
There is more (stacktrace), but I'm guessing that's not relevant.

peteruithoven, 2015-12-16:
[Zxing](https://github.com/zxing/zxing) is an open-source, multi-format 1D/2D barcode image processing library.

peteruithoven, 2015-12-16:
With my limited Java experience I can't even figure out where 51.0 is coming from, is this the version of a dependency zxing is trying use?

mgmax, 2015-12-16:
Is this specific to the Mac OS Version, or does it also happen with the zip file?

peteruithoven, 2015-12-16:
Apparently not, I can start the `VisiCut.MacOS` file in OS X. 
The last line of the error I posted (and the rest of that stacktrace) happened because the lasercutter wasn't found.

t-oster, 2015-12-16:
Could it be that the Java application stub loads Java 1.6 which is not supported because some classes need 1.7 or later?

peteruithoven, 2015-12-16:
Is there a way I could test that?

t-oster, 2015-12-17:
well you may use some kind of process inspector/task manager to see which java-executable is running. It's definitely time to #278 Fix OSX Packaging, by including a JavaRuntimeEnvironment in the app-bundle

companje, 2016-07-15:
Solution for me was to update from java 1.6 to java version 1.8 from this page: http://www.oracle.com/technetwork/java/javase/downloads/jre8-downloads-2133155.html
But after installing and rebooting it 1.6 was still the default java version. I ended up changing the `VisiCut.MacOs` shell script by referring to the absolute path of java 1.8.

```bash
#!/bin/bash
DIR="$( cd -P "$( dirname "$0" )" && pwd )"
JAVA=/Applications/Processing.app/Contents/PlugIns/jdk1.8.0_74.jdk/Contents/Home/jre/bin/java
$JAVA -Xms256m -Xmx1024m -jar "$DIR/Visicut.jar" --singleinstanceport 6543 "$@"
```

mgmax, 2019-11-01:
Closing as we don't support Java < 9 anymore.

closed 2019-11-01 by mgmax