t-oster/VisiCut #323: VisiCut crashes on linear gradients

sedrubal, 2015-12-18:
Here is a plain svg file. When opening this file, visicut crashes in a strange way... When removing the

```html
style="fill:url(#linearGradient3836);"
```

it works...

mgmax, 2015-12-18:
I can't see a file or link attached.
Could you please post a minimum example file like this one (for another issue) ?
https://gist.githubusercontent.com/mgmax/5221837/raw/25aafd208a7334adc774263c29d5a99866a136fa/clip-path-none.svg

sedrubal, 2015-12-18:
Ok, here is the content:

```svg
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:osb="http://www.openswatchbook.org/uri/2009/osb"
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:xlink="http://www.w3.org/1999/xlink"
   version="1.1"
   id="svg2">
  <defs
     id="defs4">
    <linearGradient
       y2="115.66528"
       x2="312.10812"
       y1="115.66528"
       x1="203.08"
       gradientTransform="matrix(1.3465413,0,0,1.3465413,-148.03965,-49.764813)"
       gradientUnits="userSpaceOnUse"
       id="linearGradient3817"
       xlink:href="#linearGradient3929" />
    <linearGradient
       osb:paint="solid"
       id="linearGradient3929">
      <stop
         id="stop3931"
         offset="0"
         style="stop-color:#000000;stop-opacity:1;" />
    </linearGradient>
    <linearGradient
       xlink:href="#linearGradient3929"
       id="linearGradient3836"
       gradientUnits="userSpaceOnUse"
       gradientTransform="matrix(1.3465413,0,0,1.3465413,27.34389,40.616611)"
       x1="203.08"
       y1="115.66528"
       x2="312.10812"
       y2="115.66528" />
  </defs>
  <metadata
     id="metadata7">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <g
     id="layer1">
    <path
       d="m 391.87766,205.29806 c 25.88329,-40.47811 -91.26535,-12.43885 -26.95586,3.18549 2.89052,1.57733 26.08383,15.26298 26.08383,14.77823 z"
       style="fill:url(#linearGradient3836);stroke:#000000;stroke-width:1;"
       id="path3971" />
  </g>
</svg>
```

t-oster, 2015-12-21:
This is the Exception:
````
Exception in thread "AWT-EventQueue-0" java.lang.IllegalArgumentException: User must specify at least 2 colors
	at com.kitfox.svg.batik.MultipleGradientPaint.<init>(MultipleGradientPaint.java:126)
	at com.kitfox.svg.batik.LinearGradientPaint.<init>(LinearGradientPaint.java:259)
	at com.kitfox.svg.LinearGradient.getPaint(LinearGradient.java:148)
	at com.kitfox.svg.ShapeElement.renderShape(ShapeElement.java:163)
	at com.kitfox.svg.Path.render(Path.java:92)
	at com.t_oster.visicut.model.graphicelements.svgsupport.SVGObject.render(SVGObject.java:121)
	at com.t_oster.visicut.gui.beans.PreviewPanel.renderOriginalImage(PreviewPanel.java:471)
	at com.t_oster.visicut.gui.beans.PreviewPanel.paintComponent(PreviewPanel.java:615)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JComponent.paintChildren(JComponent.java:879)
	at javax.swing.JComponent.paint(JComponent.java:1055)
	at javax.swing.JViewport.paint(JViewport.java:731)
	at javax.swing.JComponent.paintChildren(JComponent.java:879)
	at javax.swing.JComponent.paint(JComponent.java:1055)
	at javax.swing.JComponent.paintChildren(JComponent.java:879)
	at javax.swing.JComponent.paint(JComponent.java:1055)
	at javax.swing.JComponent.paintChildren(JComponent.java:879)
	at javax.swing.JComponent.paint(JComponent.java:1055)
	at javax.swing.JLayeredPane.paint(JLayeredPane.java:585)
	at javax.swing.JComponent.paintChildren(JComponent.java:879)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5230)
	at javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:295)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1249)
	at javax.swing.JComponent.paint(JComponent.java:1032)
	at java.awt.GraphicsCallback$PaintCallback.run(GraphicsCallback.java:39)
	at sun.awt.SunGraphicsCallback.runOneComponent(SunGraphicsCallback.java:78)
	at sun.awt.SunGraphicsCallback.runComponents(SunGraphicsCallback.java:115)
	at java.awt.Container.paint(Container.java:1967)
	at java.awt.Window.paint(Window.java:3877)
	at javax.swing.RepaintManager$3.run(RepaintManager.java:819)
	at javax.swing.RepaintManager$3.run(RepaintManager.java:796)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:77)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:796)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:769)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:718)
	at javax.swing.RepaintManager.access$1100(RepaintManager.java:62)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1677)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:312)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:745)
	at java.awt.EventQueue.access$300(EventQueue.java:103)
	at java.awt.EventQueue$3.run(EventQueue.java:706)
	at java.awt.EventQueue$3.run(EventQueue.java:704)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:77)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:715)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)
````

may be a bug in kitfox's SVGSalamander or batik itself. However, VisiCut will probably never undestand the full SVG standard ( #15 ).

Maybe try with an updated version of SVGSalamander (is there one?)

t-oster, 2015-12-21:
it may also be the problem that your linear gradient consists of three linked gradients...

sedrubal, 2015-12-21:
MH OK... Is it possible to ignore unknown things in svg so that visicut does not crash?

t-oster, 2015-12-22:
well we should probably wrap the call in SVGObject.java:121 in a try/catch block and just continue if there was an exception.

commit 4b7205dc (2015-12-22): Catch other Exceptions. Fixes #323
closed 2015-12-22 by t-oster