t-oster/VisiCut #325: Visicut stucks on Windows

sedrubal, 2015-12-21:
Visicut sometimes stucks on time calculation or while sending a job to the laser cutter on windows. When opening the same files on our Debian machine, everything is working fine... We are using an epilog.

sedrubal, 2015-12-21:
Ah it seems to be the same as #308

t-oster, 2015-12-21:
please try to reproduce it with starting VisiCut from the command line (or even better from netbeans with the debug button), there may be some clues in the log. Otherwise provide the exact steps to reproduce the error

sedrubal, 2015-12-21:
When I start `VisiCut.exe` in Windows from `cmd` there is no output. VisiCut starts and immediately a new prompt appears. Do you know how to do this? I have no Windows to reproduce this or to install NetBeans, sorry...

And now I couldn't reproduce this with the same file... even with some load... Before I got sometimes the message that there is not enough RAM... maybe this belongs to each other?

mgmax, 2015-12-26:
I am very certain that this bug is not Windows-specific (why should it be?) and just a duplicate of https://github.com/t-oster/VisiCut/issues/308 .

The 'not enough RAM' message happens when VisiCut hits the JVM memory limit, which is set (by `-Xmx512M`)in https://github.com/t-oster/VisiCut/blob/develop/distribute/files/VisiCut.Linux (Linux) and VisiCut.exe, generated by https://github.com/t-oster/VisiCut/blob/develop/distribute/windows/launcher.nsi

The windows version uses a lower mem limit of 512M, Linux has 1024M.

commit 0fd8a8a6 (2015-12-28): Raise memory limit for Windows to 1024M

Partly fixes #325
closed 2015-12-28 by t-oster