t-oster/VisiCut #329: Visicut and El Capitan

ToPo47877, 2016-01-14:
Hello, 
we´ve got a bunch of new iMacs in our Fablab and i tried to install Visicut on them.
Installed Java previous, installed Visicut, started, and nothing... 
The splashscreens apears, the top menu "Visicut" apears and i can get into settings. 
If i close the settings, the splashscreen disapears and the hole menu comes up, but no GUI pops up and the every menu-point is greyed out... 
Any ideas about this problem? 
I tried to get the "universal-package" to start visicut in the terminal, but the download-site says "bad gateway"?
Thanks for your help,
greets from Bottrop
Tobi

t-oster, 2016-01-14:
Hi, the download server should be online again. This thing keeps locking up, I am working on a solution

t-oster, 2016-01-14:
Please try to delete the user's .visicut folder before trying again.

ToPo47877, 2016-01-15:
Thanks! It seems that i´ve choosen some wrong setting-file on first start-up. Resetet the .visicut folder and everything is working perferct!

closed 2016-01-15 by ToPo47877