t-oster/VisiCut #330: Turnkey firmware laser won't fire

soldir, 2016-01-20:
I have the Turnkey firmware and the latest development of visicut. The laser won't fire after job was sent to it.

Also on another note.

Job won't send unless I remove SD card and then reinsert it.

mgmax, 2016-01-21:
Since VisiCut is used for many different lasercutters, your report is missing some  important information. Please post screenshots of the driver settings page and also some information on if this lasercutter works with other software. 

I have done some development work on VisiCut, but have absolutely no idea what you are referring to by "turnkey firmware". With enough information someone could make an educated guess or help you with debugging.

soldir, 2016-01-21:
It works with Inkscape plugin https://github.com/TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin 
Turnkey firmware is at this location https://github.com/TurnkeyTyranny/buildlog-lasercutter-marlin 

[image]
[image]
[image]
[image]

mgmax, 2016-01-23:
I think this are two separate problems:

1. VisiCut sends the job in a wrong way, or at least differently than 
your software. The plugin you mentioned only generates a file and does 
not send it, as far as I understood it, so how do you send the generated 
file to the lasercutter?
I don't know how to address this, since I can't compare it to the other 
software. (Using a serial port sniffer tool to look at the communication 
may help)

2. The GCode generated is wrong/different.
You can export the file from VisiCut (File - Export Laser Code) and 
compare it to a GCode created by something else, just look at both files 
with a text editor. Usually the start/end GCodes are specific to the 
firmware, so you need to adjust them in VisiCut.

Probably you should first fix the GCode settings, export file and 
somehow manually send this file to your laser.

soldir, 2016-01-25:
1. Your are correct the plugin only generates the gcode but can use Repetier to send the gcode. Ijust transfer the files to SD card and load that way.

2. The gcode that visicut is generating is different from what it is expecting. The only difference that I found  was that visicut writes S0.2000 (20%) for the laser power but turnkey does not see this as 20% but 0.2%. I changed the gcode manually to S20 and it fired like it should have. 
[image]
[image]

t-oster, 2016-02-04:
So the problem is, that the S-Values are interpreted as percent instead of fractions?

t-oster, 2016-02-04:
If this is the case for all marlin- based lasercutters, we should update the driver. However for the smoothieboard this page http://smoothieware.org/laser sais the S-Parameter goes from 0 to 1 instead of 0-100

t-oster, 2016-02-05:
@soldir please try version 1.7.9-1 from http://download.visicut.org/develop and reopen if it's not fixed.

closed 2016-02-05 by t-oster