t-oster/VisiCut #331: Laser cut "Overlap" option.

xboxhacker, 2016-01-27:
Is there a way to make the laser slightly overlap the cut end and begin points?

Should only have to be the width of the laser beam.

So as an example:
Current: Cutting a 2" circle, the laser would travel 6.28" in circumference. 
Request: Cutting a 2" circle, the laser would travel 6.295" in circumference.

I have a noticeable end "point" when cutting acrylic. A _little _more travel would eliminate it.

I am using SmoothieBoard driver.

Thank you!

t-oster, 2016-01-28:
Hi... there is currently no such option. It would generally also not be that easy, because in the circle case, the line should be round, but e.g. in a line case it should be straigt....
Of cause if you just want to hack it for you, set up an development environment https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started
and change the smoothie driver around here https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java#L319 to check:
* if the last current command is "LINE" and the next one is "MOVE"
* then manipulate the LINE coordinates to be a little longer. (i would calclulate the direction between the last point and the current MOVE and add a bit...)

t-oster, 2016-01-28:
...the other option is of caus to user Inkscape or else to modify your drawing

mgmax, 2016-01-28:
IMHO this is a hardware issue and could best be fixed in the SmoothieBoard firmware by a small pause after switching on the laser.

xboxhacker, 2016-01-30:
I think @mgmax might be correct. Have a very small dwell at the start of the cut.

@t-oster where would I find the section of your code to add custom gcode? It will not work from the "lasercutters" setting. Needs to be added after that.

`G21
G90
M3     <~~~wont work, too early
M106
G0 X21.564600 Y11.252200 F3600
~~~~~<GCODE NEEDS TO BE ADDED HERE>~~~~~~
G1 X21.513800 Y12.268200 S1.000000 F1200
G1 X21.361400 Y13.258800
G1 X21.107400 Y14.198600
G1 X20.777200 Y15.138400
G1 X20.345400 Y16.002000
...
G1 X21.513800 Y10.210800
G1 X21.564600 Y11.252200
G1 X21.564600 Y11.252200
G0 X0 Y0
M5
M107
M84
` 

Thanks!

t-oster, 2016-01-30:
See my previous comment

pelrun, 2016-02-18:
Also try increasing the tickle_power option in the smoothieboard config; this keeps the laser biased just below lasing threshold so that it turns on a lot faster than if it was coming on from zero.

computercarguy, 2016-06-27:
I like the idea of a software implemented Overlap feature.  The laser cutter software I currently use has this built in and is handy at times.

I use a variety of materials, from thick acrylic to thin acrylic, various thicknesses and types of wood, foam, Buna-N rubber, and I don't remember what all.  Each material has it's own need of an overlap.  Thick acrylic (~1/2") needs overlap while very thin stuff doesn't.  The 1.5" foam almost needs a gap, or negative overlap, since the foam melts/evaporates so easily.

I think the proper way to handle overlap is to find the intersection point with the start and follow the same path for the requested amount.  Most of the overlaps should be on the order of less than a millimeter, so this shouldn't be too significant.  Different machine had different power capabilities, so the user should be able to adjust the amount of the overlap, such as 0.5mm or 0.1mm, for different materials or patterns/projects.

For a "negative" overlap or gap that I mentioned, the laser would simply stop before it got to the end of the cut.  If it makes it easier, the beam could stop even if the head continues to move, again for less than a millimeter.

I also like the idea of a dwell, but as a software option.  I am building a laser cutter that will be capable of cutting metal, and that may require a dwell to get through some of it, while other materials, like the aforementioned foam, would not need that.

As a programmer and a user, I like the idea of having editable software features, instead of having hard coded values or "features".