t-oster/VisiCut #332: Unable to use Visicut in Win7 64

Downunder35m, 2016-02-02:
I downloaded the latest installer and all went well during the installation.
However that is all that is working for me.
Dowloading settings results in an error:
Error
Could not download settings
Error importing settings

No connection to the machine possible either, no g-code export - all gives back errors.
Even with firewall and al disabled the program is unable to connect to anything.
There was a similar issue reported earlier but with no response through several program versions.
I would like to try the program one if possible.

mgmax, 2016-02-02:
Please link to the previous issue. Reporting duplicates doesn't help.

t-oster, 2016-02-02:
Hi, please download the platform independent version https://github.com/t-oster/VisiCut/releases/download/1.7_310/VisiCut-1.7-310-gcf8c087.zip and start visicut via command line to see, what errors exactly occur.

Downunder35m, 2016-02-02:
Link to previous issue:
[https://github.com/t-oster/VisiCut/issues/251](url)
I used the Windows Installer with the same version number.
Starting from a command prompt makes no difference as there is no log output.
So I don't know how to explain it any better than: Visicut seems to be unable to connect to anything, not my machine, despite the com port showing up and working fine in Pronterface (yes, I closed pronterface).
The popup at start that asks what settings to download shows up but when the program starts there is the error I mentioned above.

So what am I suppose to see more when starting Visicut from the command prompt?

t-oster, 2016-02-03:
Please try to run VisiCut from the commandline via
```
cd path/to/your/download
java -Xms256m -Xmx1024m  -jar Visicut.jar

pelrun, 2016-02-03:
This might be the same as the issue I had with talking to a Smoothieboard via serial on Win7 64bit last night. I finally tracked it down to DTR not being set when the serial port is opened, so the Smoothieboard doesn't know to send it's initial "Smoothie" and "ok" text, and Visicut ends up timing out.I added an explicit sp.setDTR(1); line to GenericGcodeDriver.java just after the port is opened and everything started working properly.

t-oster, 2016-02-03:
This is an interesting point. I did not notice that before. How about a pull-request?

t-oster, 2016-02-04:
@Downunder35m please test with the latest version and close the issue if that was the problem.

Downunder35m, 2016-02-12:
I started work that causes me to be away from home for most of the week, sorry for the delay.
Will get the latest version and do the tests tonight and report back.
Thanks for the help!

Downunder35m, 2016-02-13:
With the above command line I can start and use Visicut.
All good I would say :)

closed 2016-02-13 by Downunder35m