t-oster/VisiCut #335: Slow upload to Smoothieboard

koswix, 2016-02-08:
I've read the related thread about Smoothieboard and saw that it ended with a request to create an issue for slow HTTP upload, but can't see that an issue has been raised. Apologies if I missed it.

Problem only seems to occur with large files (> few hundred k). I can upload via the Smoothieboard's own web UI fine (it's slow, but it does complete), but using the upload in Visicut it just hangs. 

Sending over serial (still over the network) does work, but that suffers from the related bug where Visicut is waiting for the OK response before proceeding, resulting in very jerky motion. This is noticed on vector curves and on raster in particular.

Now my previous work around was to export the laser code and upload it manually, however for engraving (2D or 3D engraving) the output file is empty of everything except the pre-job instructions, so currently I'm stuck engraving tiny files or low res 100dpi. Any ideas?

t-oster, 2017-02-23:
Hi, with sending over serial (still over the network) you mean the upload method `IP` in VisiCut's driver setting?

t-oster, 2017-02-23:
I updated the driver to not write debug output, which may have slowed down the whole writing. Please test again with 1.7.9_69_g7955704-1

mgmax, 2019-11-01:
Closing because the report refers to a very old version. Feel free to reply if the bug also happens in a recent version of VisiCut with Java >= 9.

closed 2019-11-01 by mgmax