t-oster/VisiCut #339: Problem with CorelDrawX7 SVGs in Visicut

Graphicsamateur, 2016-02-15:
Hi Guys,

I have a problem with my SVGs from CorelDraw in VisiCut. Visicut doesnt recognize the object colours as it looks. Here are my screenshots. One is from CD and one to compare from InkScape where are for this easy project also no problems.

[image]

[image]

Do you have any idea how to fix that problem?

Best greets
Alex

mgmax, 2016-02-15:
Without the files themselves I can't help. Since github doesn't allow 
attaching SVG files, please open the SVG file in a text editor, copy the 
content to pastebin.com (or github gists etc.), and send the URL.

Graphicsamateur, 2016-02-15:
Here is the file from CD

http://pastebin.com/h1kfkLdR

and InkScape if you need to compare

http://pastebin.com/DCJPp7nt

Thanks a lot

t-oster, 2016-02-15:
It looks as if Corel Draw uses CSS in a Style element inside of the defs block instead of style attributes of groups.
VisiCut uses SVGSalamander to interpret SVGs, which does not seem to support separate style blocks....

t-oster, 2016-02-15:
...or I am not using it correct, this needs further investigation.

Graphicsamateur, 2016-02-16:
So if you need any further information from my side, let me know!!???!?
It would be really nice if there would be a solution for using the Corel files!

mgmax, 2016-02-16:
Looks like the problem isn't especially easy to fix. As a workaround, you could try different export settings.

Graphicsamateur, 2016-02-16:
I tried already some different options in the Corel export settings, but all bring the same problem. 
Unfortunately Corel works the best and least problematic with my graphic files. Maybe you can do something the next days to find out some solutions?

pelrun, 2016-02-18:
What happens if you load your Corel SVG's into Inkscape and send them to Visicut using the plugin?

t-oster, 2016-02-18:
Why do you need the separate colors at all? Maybe use can use groups/layers instead of colors to classify your lines? The problem is that VisiCut does not interpret the style completely like a CSS processor, but only uses the style information directly present in the nodes.
Maybe there is also an export-setting for corel draw, which "inlines" the CSS.
If anyone wants to work on this, the code here:
https://github.com/t-oster/VisiCut/blob/develop/src/com/t_oster/visicut/model/graphicelements/svgsupport/SVGObject.java#L132
has to be changed to take the CSS definitions into account instead of just processing inline style attributes.

Graphicsamateur, 2016-02-19:
Hello!
That would be great if someone could change and fix that problem. Unfortunately I was unable to check something yesterday. I´ll try today something again with export options. But also I have to say that my Nickname here is no joke. 

@pelrun I didn´t know about this way of using VisiCut via plugin. I´ll try also this later. 

Thanks for now.

I´m still full of hope for any final solution.... :-)

mgmax, 2016-02-19:
There is also development ongoing for a Corel plugin (not sure if X7 or X6)
Still trying to push @gurkih to publish it

mgmax, 2017-04-23:
In Corel X8, you can set something like "form options = display attributes" (don't know the exact translation, in german: "Formoptionen = Darstellungsattribute"). Then VisiCut can read the files. Maybe you want to try the exercise of writing a Corel extension (macro) which calls the export with this option?

jnweiger, 2017-12-25:
I've done a minimalistic python implementation fixing this issue in inksvg.py 
https://github.com/jnweiger/inkscape-thunderlaser/commit/dffee336bf84a4a1df056973ae61eb9944356060

Maybe a Java-Wizzzard would want to re-implement that for VisiCut? It is only 50 lines of code.