t-oster/VisiCut #346: Affinity Designer SVG

emnullfuenf, 2016-03-08:
I tried to import an SVG exported from Affinity Designer into VisiCut. The file shows up in the menu but does not display. Only a handle is visible in the upper left corner.
[image]

Affinity Designer allows to set DPI and Viewbox. An A4 page with a 10mm box from Inkscape exports like this:
`<svg width="100%" height="100%" viewBox="0 0 210 297" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linejoin:round;stroke-miterlimit:1.41421;">`

The only working export settings in Affinity Designer are:
96 DPI, flatten transforms, set ViewBox disabled
Unfortunately the 100mm box is 107mm in VisiCut.
[image]
```
<svg width="794px" height="1123px" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linejoin:round;stroke-miterlimit:1.41421;">
    <rect x="0" y="0" width="377.953" height="377.953" style="fill:black;"/>
</svg>
```

Any idea? Is this Affinity's fault?

pelrun, 2016-03-08:
Definitely a bug in Inkscape; px is incorrectly defined as 90dpi. Visicut's svg handler would have inherited that. It's fixed in the development version of Inkscape, not that that helps.

You can make the svg document render correctly by forcing inkscape to use user units instead of px by changing the svg tag to include: width="210mm" height="297mm" viewBox="0 0 793.7 1122.5"

pelrun, 2016-03-08:
Just had a quick look in Visicut and it tries to determine which application generated the svg and sets the dpi appropriately. It explicitly detects Inkscape files and sets 90dpi there, but it also defaults to 90dpi if it can't tell. I think we need to change that default to 96dpi.

Files coming from Inkscape 0.92 and beyond are going to need to be distinguished as well. *sigh*

emnullfuenf, 2016-03-08:
There's a little misunderstanding. I am exporting from Affinity Designer. I just tested a document from Inkscape for comparison.

Anyway. Thanks for the hint. I thought Inkscape works with 96dpi. When i am exporting the 100mm box from Affinity Designer at 90dpi it ends up in Visicut at the right size.

Finally the right settings for the SVG export in Affinity Designer are:
**90 dpi
flatten transforms
set ViewBox disabled**

ventilator, 2019-08-12:
@emnullfuenf 

I am also using Visicut and Affinity Designer. Sometimes fx Effects do not show up in Visicut or even Visicut crashes. Do you happen to know a workaround how to Export SVGs with Affinity Designer, so Visicut can understand them? Opening and saving the SVG with inkskape did not solve the issue.

emnullfuenf, 2019-08-12:
I had similar problems in the last weeks, too. But i was in a hurry so i couldn't search for a stable solution.

What kind of fx are you trying to cut?

ventilator, 2019-08-12:
Just a simple color overly (=making things black for engraving) - but only the red rectangle is imported into visicut, the shape (the leave) is not visible
[image]

[3signs_cccamp19.zip](https://github.com/t-oster/VisiCut/files/3492714/3signs_cccamp19.zip)

emnullfuenf, 2019-08-12:
With the above settings everything worked in Visicut 1.7-295-g4f907b8. Exported from Affinity Designer 1.7.1.

Have you checked these settings?
90 dpi
flatten transforms
set ViewBox disabled

ventilator, 2019-08-12:
That is strange, I am on VisiCut 1.8, Affinity Designer 1.7.1 with said settings. This generates this file, which then is imported into Visicut. Can you give me your svg to crosscheck my setup? Thanks.
[3signs_cccamp19_fx_test.svg.zip](https://github.com/t-oster/VisiCut/files/3492947/3signs_cccamp19_fx_test.svg.zip)

emnullfuenf, 2019-08-12:
My version only shows in Visicut the second time. I have to import it twice via the "File:" icon.

[3signs_cccamp19__.svg.zip](https://github.com/t-oster/VisiCut/files/3493134/3signs_cccamp19__.svg.zip)

ventilator, 2019-08-12:
Hm, my file contains black rockets (the image), but rasterized. Your files contains colored rockets, vectorised. Thus they are imported correctly, but with different colors, so I would have to define multiple colors for engraving. It seems it is just stripped from the fx effect to make them black. Screenshot shows your file
[image]

Anyway, the simple solution is to just make them black by changing the stroke color from colored to black and leave the fx effect out. but in another case it might be usefull to have the option for fx effects. Thanks for your support

icemansparks, 2021-07-08:
> There's a little misunderstanding. I am exporting from Affinity Designer. I just tested a document from Inkscape for comparison.
>
> Anyway. Thanks for the hint. I thought Inkscape works with 96dpi. [...]

Five years later and this is still the only working solution, unbelievable, but thank you very much for the details!

t-oster, 2021-07-08:
@icemansparks it's not so unbelievable if you keep in mind that this issue was closed in 2016 and visit is entirely developed by volunteers, who probably have a life and a job and may not even use/know Affinity Designer. Feel free to provide examples, detailed bug reports or even patches.

icemansparks, 2021-07-09:
> @icemansparks it's not so unbelievable if you keep in mind that this issue was closed in 2016 and visit is entirely developed by volunteers, who probably have a life and a job and may not even use/know Affinity Designer. Feel free to provide examples, detailed bug reports or even patches.

@t-oster please do not take this as offense, i was just impressed, that such a problem (in regards to affinity designer) still exists, since it impacts me on more levels than just the visicut usage. Thanks for the great work here!

t-oster, 2021-07-09:
@icemansparks so the issue is not with visicut's way of interpreting SVGs but with Affinity Designer creating them?

icemansparks, 2021-07-09:
> @icemansparks so the issue is not with visicut's way of interpreting SVGs but with Affinity Designer creating them?

Yes, exactly - it is "nearly impossible" to export from affinity designer in a way that the file can be used in other programs (e.g. Illustrator) without any hastle.

t-oster, 2021-07-09:
@icemansparks ok, than sorry for the missunderstanding. Maybe we could link the bug report in Affinity Designer here so people know where to complain (is there a bug report?)

mgmax, 2021-07-10:
Possible solutions for the resolution issue:

- ask Affinity to set the correct width/height in mm (not px) and add a viewbox attribute by default.
- create an Export plugin for Affinity (not possible, they don't have any macro/plugin support)
- detect Affinity-created SVGs and work around, just like we do for Illustrator and old Inkscape: https://github.com/t-oster/VisiCut/blob/9e819d34b501d015cc9fe7bc214ca3516db410b7/src/main/java/de/thomas_oster/visicut/model/graphicelements/svgsupport/SVGImporter.java#L189
The detection should be possible by the line `<svg width="1234px" height="1234px" ... xmlns:serif="http://www.serif.com/" ...`. The difficulty is that the workaround should not break things once Affinity have fixed their export. Maybe check that there is no viewBox?