t-oster/VisiCut #349: all-in-one IOT base CNC machine

mkeyno, 2016-03-22:
hi @t-oster, I was wonder have you check the[ LaserWeb](https://github.com/openhardwarecoza/LaserWeb/) front-end CAM host of @openhardwarecoza  and [serial bridge of](https://github.com/luc-github/ESP8266) @luc-github , the plan is all the CAM host embedded in the ESP Flash and such as IOT device  just connect the printer and start design and printing (cutting ), I was hopping have your comments about that 
BR

peteruithoven, 2016-03-22:
@pelrun It was mentioned here: https://github.com/t-oster/VisiCut/issues/232#issuecomment-168847105
Very interesting project! 
But my hope is that it will get more user friendly. It seems to have a lot of functionality, but for more novice users, like Fablab visitors it's probably a bit to much. This is mostly about progressive disclosure;
http://seanrice.net/user-experience/patterns/progressive/disclosure/ux/2013/09/26/progressive-disclosure.html
https://en.wikipedia.org/wiki/Progressive_disclosure
I can't imagine a Fablab user needing for example the console, seeing the x, y, z position, seeing the content of a job etc. So these could start out closed / folded. A button to enable 3D printer mode could be something that's configured in the specific machine settings. 
See for example a bit of Visicut's progress in the following issue: https://github.com/t-oster/VisiCut/issues/68

openhardwarecoza, 2016-03-22:
> @pelrun <https://github.com/pelrun> It was mentioned here: #232 (comment)

See laserweb2.0 please,  instead of continuing to point users at codename
clowncar which was laserweb1 (;

mkeyno, 2016-03-22:
wow interesting activity on web UI out there @peteruithoven  @pelrun , but it seems no body mention the capability the IOT to gather all needs along with all easiness , even @openhardwarecoza has not embedded its CAM host inside the wifi Node links, 
I was hopping in here, web programming talented guys can join with embedded wifi Node host developer to links this last chain

peteruithoven, 2016-03-22:
@mkeyno Could you eleborate on those wifi Node links? Are those the hardware that links the web interfaces to the lasercutters (through usb etc)? 
@openhardwarecoza are you saying [laserweb](https://github.com/openhardwarecoza/LaserWeb/) is "depricated" and we should look at the following [LaserWeb2](https://github.com/openhardwarecoza/LaserWeb2)?

pelrun, 2016-03-22:
Dunno why I'm mentioned here...

Anyway, this really has little or nothing to do with VisiCut.

openhardwarecoza, 2016-03-22:
Busy porting over the functionality but yes,  laserweb2 is where the action
is.  Cleaner UI. The community affectionately calls LW1 the 'clown car
interface'    also,  no nodejs dependency.  Just waiting on Luc or others
to add websocket support to esp8266 serial bridge then you literally can
compile lw2 into a single html file and put it on a esp8266 if you wanted
to.  (community wants to,  i dont care either way,  its a web page.  Host
it wherever you want (cloud,  local,  esp whatever).   The js for said
websocket talking isnt done yet since there isnt a websocket implementation
on a esp8266 yet. OP keeps tagging me in these posts,  so sorry,  no
competition or offence meant.  Remember laserweb is just an app I need.
But the community seems to so more with it that i anticipated hehe.

pelrun, 2016-03-22:
Can we close this please? Issues are not a discussion forum.

openhardwarecoza, 2016-03-22:
Fine with me.  @mehrdad, feel free to come discuss it on laserweb's issues
page (;  but please try to not offend other projects with these unrelated
posts (;

mkeyno, 2016-03-22:
sorry for unintentional trouble for anyone here , I just thinking how would be great to have comments of  talented people on each other effort  that may unaware , 
however I'm not aware how to make  discussion forum and yes I will close this issue 
also I'm sorry peter @openhardwarecoza to mention you many times , but believe that I'm so totally believe to what you have done yet
BR to all

closed 2016-03-22 by mkeyno

pelrun, 2016-03-22:
Sorry, I didn't want to be harsh, it's just a discussion that would be much more usefully held elsewhere.