t-oster/VisiCut #350: Add Update-Check and In-App Update

t-oster, 2016-03-24:
Since we are releasing quite often, it should be more easy for the users to follow the development, so I would recommend adding an automated/manual update check in the app with optional downloading of the latest version.

mgmax, 2024-03-03:
For traditional installations (Windows setup, Linux .DEB package, maybe Mac), this would be quite easy. Just download the new file, execute it and restart VisiCut.

For the portable ZIP and especially the AppImage it may be quite hard, as it is not really clear what is the expected behavior. Overwrite the old file? Download the new file and let the user do the job?

TheAssassin, 2024-03-03:
AppImages expose this functionality as well through AppImageUpdate.

Also, one could just ask the GitHub API for new releases. But beware that there is some rate limiting there, too. I have an idea how to circumvent that, though.