t-oster/VisiCut #351: Linux patch

z4ziggy, 2016-03-27:
the following is needed to have VisiCut working on Linux:

```
diff --git a/src/com/t_oster/visicut/gui/MainView.java b/src/com/t_oster/visicut/gui/MainView.java
index 831f494..eee5b89 100644
--- a/src/com/t_oster/visicut/gui/MainView.java
+++ b/src/com/t_oster/visicut/gui/MainView.java
@@ -1896,7 +1896,7 @@ public class MainView extends javax.swing.JFrame
   {
     final FileFilter allFilter = VisicutModel.getInstance().getAllFileFilter();
     //On Mac os, awt.FileDialog looks more native
-    if (Helper.isMacOS())
+    if (Helper.isMacOS() || Helper.isLinux())
     {
       FileDialog openFileChooser = new FileDialog(this, bundle.getString("PLEASE SELECT A FILE"));
       openFileChooser.setMode(FileDialog.LOAD);

```

mgmax, 2016-03-27:
Can you please describe that in more detail, possibly with screenshots? 
It works for me without this change.

z4ziggy, 2016-03-27:
without the patch:

[image]

with the patch:

[image]

t-oster, 2016-03-27:
Which distribution/window Manager/Toolkit/Java Version are you using? Some use the native file requester with FileDialog and some look better with JFileChooser. There is even a gtk specific implementation (lib/gtkfilechooser)

mgmax, 2016-03-27:
Okay, your request was not very clear. It is actually a duplicate of #317, so we should continue discussion there. It looks like your patch is the solution at least for Linux+Gnome.

t-oster, 2016-03-27:
yes, it is a duplicate of #317 so I close this. @z4ziggy please post your screenshots there together with exact distro/desktop environment/java-version. Maybe people with other constellations can test this too

closed 2016-03-27 by t-oster