t-oster/VisiCut #352: Visicut fails to launch via exe under Windows

grbd, 2016-03-30:
When launching visicut from windows 10 via the .exe this seems to be currently broken

1. With the latest version 1.7.9.30
Launching the exe fails with the below error
Launching the jar file directly seems to work

2. With an older version 1.7.310
Launching the exe works but uses a really old looking window frame
Launching the jar file directly seems to work best

I've attached the error log below, although I'm not sure if it's much use

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x7899ac5e, pid=15992, tid=5496
#
# JRE version: Java(TM) SE Runtime Environment (8.0_77-b03) (build 1.8.0_77-b03)
# Java VM: Java HotSpot(TM) Client VM (25.77-b03 mixed mode windows-x86 )
# Problematic frame:
# C  [AcGenral.DLL+0x3ac5e]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x06366800):  JavaThread "Swing-Shell" daemon [_thread_in_native, id=5496, stack(0x08660000,0x086b0000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x0974c081

Registers:
EAX=0x09b10000, EBX=0x00000000, ECX=0x09710000, EDX=0x004de000
ESP=0x086aea58, EBP=0x086aea60, ESI=0x00ae0000, EDI=0x0974c081
EIP=0x7899ac5e, EFLAGS=0x00210287

Top of Stack: (sp=0x086aea58)
0x086aea58:   00ae0000 0974c081 086aea78 7899a94f
0x086aea68:   00000000 00ae0000 00000000 086aeaac
0x086aea78:   086aea98 789bbb26 0974c085 761f9fc9
0x086aea88:   15feddc8 761fe3e5 00ae0000 00000000
0x086aea98:   086aeab4 789bb2d5 0974c085 086aeaac
0x086aeaa8:   15feddc8 00000000 1b33dd1d 086aead4
0x086aeab8:   755e4e63 00ae0000 00000000 0974c085
0x086aeac8:   800106f8 00000000 75828eb4 086aeb7c 

Instructions: (pc=0x7899ac5e)
0x7899ac3e:   15 d4 30 a1 78 85 ff 74 3b 8b ce 3b f9 76 08 8b
0x7899ac4e:   01 03 c1 3b f8 72 09 8b 49 04 85 c9 75 ed eb 0e
0x7899ac5e:   8b 07 25 01 00 00 f0 3d 00 00 00 a0 74 16 6a 57
0x7899ac6e:   ff 15 d0 30 a1 78 8b 55 08 8b ce e8 7d fe ff ff 

Register to memory mapping:

EAX=0x09b10000 is an unknown value
EBX=0x00000000 is an unknown value
ECX=0x09710000 is an unknown value
EDX=0x004de000 is an unknown value
ESP=0x086aea58 is pointing into the stack for thread: 0x06366800
EBP=0x086aea60 is pointing into the stack for thread: 0x06366800
ESI=0x00ae0000 is an unknown value
EDI=0x0974c081 is an unknown value

Stack: [0x08660000,0x086b0000],  sp=0x086aea58,  free space=314k
[... 245 lines cut ...]
VM Arguments:
jvm_args: -Xms256m -Xmx1024m 
java_command: Visicut.jar --singleinstanceport 6543
java_class_path (initial): Visicut.jar
Launcher Type: SUN_STANDARD

Environment Variables:
PATH=C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\Microsoft\Windows Azure Tools\Docker\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Qt\Qt5.5.1\5.5\mingw492_32\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\ManagementStudio\;C:\Program Files (x86)\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Git\cmd;C:\Program Files\Perforce\DVCS\;C:\Program Files\doxygen\bin;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Windows\SysWOW64\WindowsPowerShell\v1.0\Modules\TShell\TShell\;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files (x86)\VisiCut;C:\Program Files\OpenVPN\bin;C:\Python27;C:\Python27\Lib\site-packages\;C:\Python27\Scripts\;C:\Program Files (x86)\CMake\bin;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\ProgramData\chocolatey\bin;C:\Users\richard\.dnx\runtimes\dnx-clr-win-x86.1.0.0-rc1-update1\bin;C:\Users\richard\.dnx\bin;C:\Program Files (x86)\Nmap;C:\altera\15.1\modelsim_ase\win32aloem;C:\altera\15.1\modelsim_ae\win32aloem;C:\Program Files\Docker Toolbox
USERNAME=richard
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 42 Stepping 7, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 10.0 , 64 bit Build 10586 (10.0.10586.0)

CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 42 stepping 7, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, clmul, ht, tsc, tscinvbit

Memory: 4k page, physical 6193200k(1230516k free), swap 14520368k(6140156k free)

vm_info: Java HotSpot(TM) Client VM (25.77-b03) for windows-x86 JRE (1.8.0_77-b03), built on Mar 20 2016 22:00:03 by "java_re" with MS VC++ 10.0 (VS2010)

time: Wed Mar 30 19:37:07 2016
elapsed time: 1 seconds (0d 0h 0m 1s)
```

pelrun, 2016-04-25:
I haven't had any luck replicating this - it works fine on my win10 dev machine. Would like to hear if anyone else has had a similar problem.

A segfault like this almost makes me suspect a problem with the Java installation.

mgmax, 2016-05-07:
This must be a problem with your JRE8. Have you tried launching Visicut with your JRE8 java by hand?
Use ``java -version` to see which java program you have called, and `java -Xms256m -Xmx1024m Visicut.jar --singleinstanceport 6543` to launch VisiCut just like the launcher did. If this still fails, it is definitely a problem with your java installation.

mgmax, 2017-08-28:
@t-oster The original commenter doesn't respond and nobody could reproduce the error -- Could you please close the issue?

grbd, 2017-08-28:
Sorry I'd forgoten about this one, it appears to be fine now with the latest version

closed 2017-08-28 by grbd