t-oster/VisiCut #353: Visicut fails on Execute with Engrave Solid on Manchester Hacspace Orange Laser Cutter

grbd, 2016-03-30:
1. Select the Manchester "Manchester Hacspace Orange Laser Cutter"
2. Under the profile settings select "Engrave Solid"
3. Click Execute

For the latest Visicut 1.7.9.30 and any of the 1.7.9 branch 
This results in the error
```
Error
Exception: Unknown setting 'engrave bottom up'
```

For Visicut 1.7.310 this seems to work fine without problems

TheGuv, 2016-04-15:
The current 1.7.1 stable version also fails for me on this when the 'focus is initially on the base...' box is ticked (of course, the laser has to support Z axis for this) and the job include engraving.

TheGuv, 2016-04-15:
and 1.7-310-gcf8c087-1_all.deb also fails in the same way. Curses, why didn't I backup the version that was working....

t-oster, 2016-04-15:
If you're a Developer, please git bisect in order to find the changing commit.i guess its just an incompatibility for the old laserprofile XML File. Just try to create a new 'engrave' Mapping

pelrun, 2016-04-15:
https://github.com/t-oster/LibLaserCut/commit/13b1e7763180f711f7e6f876c3a97db93ab4e1d4#diff-98b73c710e88d2bd83ae6d7a70d9fea0R42

It's at this point in libLaserCut. The Laos driver subclasses LaserProperty to add support for that engrave option, but it doesn't clone properly.

mgmax, 2016-04-16:
I tried to understand how to fix the problem, but it isn't obvious to me. @madleech could you try to help out?

closed 2016-04-18 by t-oster