t-oster/VisiCut #359: IO Error when trying to send job to Smoothieboard

tealvince, 2016-04-18:
Not sure if this is a genuine issue or simply an expected incompatibility with my obsolete system (WinXP), but when trying to write to a smoothieboard, the job quits with the following error.  Note that VisiCut appears to recognize a response from the board, and the connection can be separately verified by sending GCode commands through COM5: in Pronterface.

-- Console output follows --
< Smoothie
< ok
java.io.IOException: IO Error from COM5: Underlying input stream returned zero b
ytes
        at com.t_oster.liblasercut.drivers.GenericGcodeDriver.connect(GenericGco
deDriver.java:730)
        at com.t_oster.liblasercut.drivers.GenericGcodeDriver.sendJob(GenericGco
deDriver.java:804)
        at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:754)
        at com.t_oster.visicut.gui.MainView$64.run(MainView.java:2185)

t-oster, 2016-04-18:
try to increase the timeout

tealvince, 2016-04-20:
Interesting.  The log still shows "Smoothie...ok" as soon as I click on "Execute", and increasing the "milliseconds to wait for response" setting just increases the time delay before I get the "Underlying input stream returned zero bytes" error afterwards.

Is "Smoothie...ok" is a response coming from the Smoothie board?

Thanks.

tealvince, 2016-04-22:
Weird.  I've tried the same from my Macbook and get similar results.  I've set up a laser cutter using the smoothie driver with a serial connection, and when I tap "execute", nothing happens and 15 seconds later I get the message:

Exception: IO Error from /dev/tty.usbmodem1411: Underlying input stream returned zero bytes

Yet, if I open up a serial connection manual from a terminal (e.g. screen /dev/tty.usbmodem1411 50000), I can communicate with the laser cutter and it responds to GCode commands both querying its state and moving the head.

tealvince, 2016-04-22:
Ok.  I found the problem.  In smoothieboard.java the getter getIdentificationLine() checks for the value of the host field, but since this value is filled in by default, it is typically non-empty even if the uploadMethod is set to serial.  I have a couple of suggestions:

1) Check the upload method in getIdentificationLine, e.g.:

  public String getIdentificationLine()
  {
    if (getHost() == null || "".equals(getHost()) || UPLOAD_METHOD_SERIAL.equals(this.uploadMethod))
....

2) waitForIdentificationLine threw an exception when it timed out, so it never showed the error message indicating that it was waiting for the wrong identification line.  I suggest catching the exception for a timeout in waitForIdentificationLine so the right message can be shown.

3) I found it helpful to expose the baud rate for smoothie connections since this is configurable on smoothie boards

4) Although it turned out to be unnecessary, I added RTS/CTS and XON/XOFF support, though this turned out not to be needed.

tealvince, 2016-04-22:
FWIW, (I've done a little Java) here is a diff of changes I've made:

[vince_smoothie_serial_diff.txt](https://github.com/t-oster/VisiCut/files/231121/vince_smoothie_serial_diff.txt)

t-oster, 2016-04-22:
Hi thank you for tracking that down. Why don't you send a pull request? But I would leave out the xon/xoff.

tealvince, 2016-04-22:
Sure thing.  How do I do that?  (I've never been on github before).

pelrun, 2016-04-22:
Basically you fork the repo (you want t-oster/LibLaserCut, not the VisiCut repo) to your account, make your changes and push them to a branch on that fork, then you get a pull request button for those changes on the website. https://help.github.com/categories/collaborating-on-projects-using-issues-and-pull-requests/ goes into it in detail.

Anyway, this is a simple change and since it's my fault for breaking it anyway, I can pop in the fix.

closed 2016-04-22 by t-oster

pelrun, 2016-04-22:
Found a similar issue with the IP upload method as well, so they should be both fixed now.

There's no need to change the baud rate on the smoothieboard, the usb driver ignores it and uses the fastest speed anyway.