t-oster/VisiCut #361: Inkscape plugin does not run or install under Windows

tealvince, 2016-04-23:
I ran into a couple of problems withe the Inkscape plugin under Windows (XP).  I've included the version I hacked to make it work (FWIW, as I've never programmed in python before):

- When the corresponding menu item is selected, Visicut put up a message saying it installed the plugins properly, but the files didn't seem to be copied over for some reason, so I copied them over manually into the extension folder.

- When running the plugin, Wow64DisableWow64FsRedirection threw an attributeException (not found), so I wrapped it in a try/catch to get the script to run.

- visicut_export.py looks for the wrong executable (visicut.exe instead of visicut.jar), so I modified "which()" to take an optional ".jar" file extension, and then added extra arguments at the front of the command to launch the java runtime ("abc/def/java.exe -jar")

[visicut_export.txt](https://github.com/t-oster/VisiCut/files/232818/visicut_export.txt)

pelrun, 2016-04-23:
Is your Inkscape installed somewhere other than C:\Program Files\Inkscape?

pelrun, 2016-04-23:
Oh no, in XP Visicut tries to install the extension to the .inkscape/extensions folder in your user directory, not the main install directory. Can you check there to see if it's present?

mgmax, 2016-04-23:
Thank you for your report. I assume you are using the "platform 
independent ZIP" and not the windows setup file. It worked fine for me 
with Win7 and the setup EXE file last time I tried, so the problem is 
more complicated.

It looks like there are multiple separate bugs here. Please try to 
address them separately and create individual pull requests for the 
solutions. Here are my best guesses:

1. Windows Wow64Redirection crap exception

  -> adding the try-catch looks correct. Testing on Win7 would be good.

2. "Platform independent ZIP" doesn't contain EXE file

  -> build script needs to be changed to include this file in the ZIP

Your suggested workaround would break the plugin for all other Windows 
users with version 7/8 of the JRE or other java install directories. 
That's why at least the setup supplies a visicut.exe file which figures 
out how to launch Visicut properly.

3. Wrong plugin installation directory on WinXP (?)

  -> more testing is necessary to see whether the problem only happens 
with WinXP.
As it worked with Win7 and WinServer2008 the last time I tested, I am 
not really sure what could be the problem. But Microsoft never ceases to 
surprise you :-)

There is also a difference between 32-bit and 64-bit versions of each 
Windows version. Additional confusion comes from Inkscape, which 
switched from 32-bit to 64-bit at about version 0.91.

tealvince, 2016-04-23:
Yes, the extension files are copied to mydocuments\vince\.inkscape\extensions .  Perhaps it should ask the user where to install the files?  

Beyond that, the script itself needed some modification as it is trying to find and run visicut.exe file instead of running java.exe and passing it visicut.jar.  I don't know if perhaps the windows installer includes some sort of stub exe file for this, but as I am running XP (which the installer doesn't support), I used the universal install from the .zip file which only has the .jar file.

pelrun, 2016-04-23:
There's XP-specific code to install the extensions to a specific location. If Inkscape has changed the location of it's user prefs dir that would explain why it's not working; we just need to update it.

The launcher exe is really not that important; it just tries a couple of methods to find the java executable (in fact it's a bit broken; I have an open issue to fix the JRE detection in the installer and the launcher.) It's a useful convenience, but there's no reason we need to rely on it for internal invocations.

mgmax, 2016-04-23:
Because the inkscape plugin is not the only one - there's one for 
Illustrator and another one for corel is under development - we should 
delegate as much as possible to the launcher so that we don't have to 
repeat the code for every new plugin.

mgmax, 2016-04-23:
for reference, the XP specific code is here: https://github.com/t-oster/VisiCut/blob/eb16aabb4cc195a378b3e9e256b5a53da9c99194/src/com/t_oster/visicut/misc/Helper.java#L257

As far as I remember, some ancient Inkscape versions together with WinXP actually used the .inkscape directory. We might break something for certain users if we change VisiCut's behaviour. @tealvince is there a specific reason why you need to use WinXP?

pelrun, 2016-04-25:
Found the inkscape preferences path logic, and it's annoyingly complicated: https://github.com/inkscape/inkscape/blob/870e925655867eef852fc2a670f4f04a4af6515f/inkscape-launchpad/src/inkscape.cpp#L1230

tealvince, 2016-04-26:
@mgmax, I'm running XP only because my home PC is ancient and is only used for laser cutting and 3D printing and therefore hasn't warranted replacement (I use Macs at work).  I understand that XP isn't a supported OS anymore, but was happy to find out that I could get Visicut to work on it with only minor changes.  It works quite well now, BTW.  If it's too complex to support automatically, just documentation for other XP users would be helpful.

mgmax, 2016-04-26:
If the problem is fixed, could you please close the issue (or otherwise state what exactly hasn't been fixed yet)?

tealvince, 2016-04-26:
Sorry for the confusion.  I fixed the problem on my installation only by manually installing the inkscape extension and editing the extension python file, but AFAIK the following issues probably still need to be addressed to work for everyone else:

1) Visicut.exe (the launcher?) is not present in the universal install zip file, so the inkscape plugin doesn't function if installed from the zip file.

2) The inkscape plugin fails with a Wow64DisableWow64FsRedirection not found exception for me, and wrapping it in a try/catch (which is what I did on my copy) fixes it.

3) Visicut installed the plugin in the wrong directory for me (MyDocuments/.inkscape/extensions instead of myPrograms/Inkscape/share/extensions).  It created the .inkscape directory as part of the installation process, but there are no other files there other than what it installed.  Separate from the issue of detecting the correct path, IMHO it could be improved with fallback behavior that puts up a warning if trying to install to a non-existent directory and giving the user an option to manually select an alternative directory instead.

mgmax, 2016-04-26:
I think 1) is yet to be fixed, but the rest should already be solved in the very latest development version.

PR #364 (2016-04-30): include Windows launcher in distributed ZIP file
- include the launcher VisiCut.exe in the platform independent ZIP
  - Inkscape plugin installation now works when launching Visicut from the extracted ZIP without the setup (tested on Win7 with Inkscape 0.91). Fixes #361
  - To simplify the code, the windows installer is always built if NSIS is found.
- remove two unneccesary example files that had a name collision for non-case-aware OS (Epilog ZING vs Epilog Zing)
commit 582c8745 (2016-04-30): include Windows launcher in distributed ZIP file

- include the launcher VisiCut.exe in the platform independent ZIP
-> Inkscape plugin installation now works when launching Visicut from the extracted ZIP without the setup (tested on Win7 with Inkscape 0.91). Fixes #361
-> To simplify the code, the windows installer is always built if NSIS is found.
- remove two unneccesary example files that had a name collision for non-case-aware OS (Epilog ZING vs Epilog Zing)
closed 2016-04-30 by t-oster